LibrePCB/LibrePCB #202: why aligement up move downwards ?while center is not center

243750496, 2017-11-30:
[image]

[image]
after setting axis the position is correct now 
[image]
but how strange it is that set right move left??

243750496, 2017-11-30:
New Q:do that means choos right equal to :
Position :right middle side of "#Name"
compare with :center of the map
then do alignment: align the right middle side of "#Name" to the center of the ground

am i right

ubruhin, 2017-11-30:
> am i right?

Yes. The alignment defines where the origin (the thin cross) of the text is located.

@243750496 Please use [our IRC channel](https://webchat.freenode.net/?channels=#librepcb) for questions - We use GitHub issues for bug tracking, not for providing support.

closed 2017-11-30 by ubruhin