LibrePCB/LibrePCB #204: the category does not exist in library data base

243750496, 2017-11-30:
[image]

what happend ?do it can be able to use without cause problem?how to fix it?

ubruhin, 2017-11-30:
This is a bug in the library editor which is quite easy to fix :) Someone interested in creating a pull request?

Hint: Look at the console output to see where the error occurs, set a breakpoint there and go up the callstack :wink:

rnestler, 2017-12-13:
@ubruhin Did you assign me to mentor the issue or to fix it?

dbrgn, 2017-12-13:
FIXIT FIXIT FIXIT!!!11!11eleven

ubruhin, 2017-12-13:
Of course to fix it. I though this issue perfectly suits your skills :rofl:

PR #248 (2018-04-06): LibraryEditor: Fix fetching of package categories
Fixes #204
commit 3593adde (2018-04-06): LibraryEditor: Fix fetching of package categories (#248)

Fixes #204
closed 2018-04-06 by ubruhin