LibrePCB/LibrePCB #205: Create documentation for users

ubruhin, 2017-11-30:
We really need to start creating documentation for users. But what documentation system should we use?

Possible requirements:
- Online version (e.g. hosted at docs.librepcb.org)
- Offline version (e.g. *.pdf)
- Multilingual support
- Simple source format (human readable, ability to use version control system)

dbrgn, 2017-11-30:
Do you think a PDF would really be used?

If you use [Sphinx](http://www.sphinx-doc.org/en/stable/) you'll get both HTML and PDF docs. I also like [Gitbook](https://www.gitbook.com/) and [mkdocs](http://www.mkdocs.org/) though.

I'm a bit skeptical about multilingual, since different languages need to be kept in sync. English and German might be feasible though. But I think in that case we could just create two copies of the same docs, without any language switching functionality.

By the way, don't forget to add as much in-program docs and hints as possible, so that people don't need to go to the documentation in the first place! (For things like the library format, a prose introduction will probably be necessary though, to introduce the concepts.)

ubruhin, 2017-12-01:
> Do you think a PDF would really be used?

I think there are many users which like to have a documentation bundled with the application itself. It's not an absolutely must-have for me, but it would be a big plus if PDF export is possible.

> I'm a bit skeptical about multilingual, since different languages need to be kept in sync.

Of course, but IMO free Software should at least allow the community to translate the English documentation in their native language.

> By the way, don't forget to add as much in-program docs and hints as possible, so that people don't need to go to the documentation in the first place!

I agree, but we can only embed very few documentation into the application without clutter up the GUI. I think in many cases a link to the documentation would be a reasonable way to go.

Btw, this is quite interesting: https://github.com/KiCad/kicad-doc/blob/master/doc_alternatives/README.adoc

dbrgn, 2017-12-01:
Hm, it looks like Gitbook actually [supports Asciidoc](https://toolchain.gitbook.com/syntax/asciidoc.html), [multilingual books](https://toolchain.gitbook.com/languages.html) and [generating PDFs and eBooks](https://toolchain.gitbook.com/ebook.html).

Regarding multilingual stuff, I'm not sure how the translation workflow would work though. I'm not sure whether you can use gettext for translation (which makes it possible to easily use tools like Transifex). But on the other hand, translation through Github pull requests would probably be fine :)

Edit: Hahaha, seems I had that question already [back in 2014](https://github.com/GitbookIO/gitbook/issues/378) and forgot 😂

ubruhin, 2017-12-03:
I have created a separate repository for the documentation: https://github.com/LibrePCB/librepcb-doc

The documentation is published using GitBook: https://librepcb.gitbooks.io/librepcb/content/en/

I'm not (yet) really confident with GitBook. But at least now we have a tool for creating and publishing documentation :wink: And with using Asciidoc as file format it should be easily possible to switch to another tool some time...

ubruhin, 2017-12-04:
Now we even have a short URL: https://docs.librepcb.org/en/ :smiley:

ubruhin, 2017-12-19:
I added a Getting Started Guide to the documentation: https://docs.librepcb.org/getting_started/

So I think we can close this issue now :smiley:

closed 2017-12-19 by ubruhin