LibrePCB/LibrePCB #208: the library elements can not be saved because is not valid

243750496, 2017-12-01:
[image]
how to reproduce the bug:
1:create a symbol and draw the elements 
2.creat the component and save
3.reedit the symbol
4. update the library
5.save

243750496, 2017-12-01:
maybe 2 GND lead to the problem???
[image]
**how to fix it?**

243750496, 2017-12-01:
plese add link function , here is a test 
[image]
[image]

243750496, 2017-12-01:
is it useful to difference the GND Mark???????? 
i think it's also ok to allow us to set normal mark not with number to difference it

[image]
no i have able to save now

ubruhin, 2017-12-01:
This is a problem of how invalid objects are handled, which is discussed in #157.

closed 2017-12-01 by ubruhin