LibrePCB/LibrePCB #209: Please add 2 function: copy elements & copy text

243750496, 2017-12-01:
i can not paste(**while type ctrl+v nothing happened**) and copy elements to get a quick start of project
if each time i will do the samething it will be waste of time and boring
i hope you can fix it 
thx a lot
[image]

243750496, 2017-12-01:
[image]
1.if i want to make a matrix without copy and paste like now , it would be a vary difficult working ,
2.and if i can't add a text or show up the change the name  or value (it will not show up like i said now ) , i will easily get puzzled and difficult to compare with what i designed (missing some key or not)

243750496, 2017-12-01:
btw: i add changed the Text to Axoplast and the name to #NAME 
like this
[image]

but still nothing happend i think it's your fault that you do not create a dynamic link to the Symbol 
please fix it
[image]

243750496, 2017-12-01:
if i delete the component then recreated the component in database updated but you will see a **bug :**
that database changed while using it there's no change
[image]

243750496, 2017-12-01:
though i changed the name to #Name of the Symbol name it still show up an old one with no change(both component and already added components ) 
[image]

243750496, 2017-12-01:
BTW: is it also a bug that the one should be 1 used the 2 instead?(in image there have two 2 , as you can see), maybe it's a bug signal?
[image]

243750496, 2017-12-01:
all i have now:
[image]

ubruhin, 2017-12-01:
@243750496 Thank you very much for testing LibrePCB and reporting bugs. But unfortunately you do that in a very impolite way. Statements like "**still have bug !!!!!!!!!!!!!!!!!!!!!!!!!!!!**", "**please fix it**" or "**it's a bug!**" make me really angry. In addition, you really mess up the issues, create comments on issues which do not have any relation and so on. We cannot work that way, we need very much time to tidy up everything. We're not a company, we don't have a support department, we don't get paid for that work (except some donations).

I also already wrote you two times that this is not a support forum, but you still use it like that. Please use our IRC channel for support questions. I'm really sorry but I decided to prevent you from creating new issues/comments. You can still contact us in our IRC chat.

Thanks.

closed 2017-12-01 by ubruhin

dbrgn, 2017-12-01:
@243750496 you can find our chat here: https://webchat.freenode.net/?channels=#librepcb (or with any IRC client connected to #librepcb on freenode). it's the better place to ask for support than on github.

a lot of things simply aren't quite finished yet. if you want to use librepcb at this stage you'll need some patience and/or willingness to contribute code fixes :) we hope to be able to publish a first preview release in the next few months.