LibrePCB/LibrePCB #212: Useless animations when selecting footprints

wavexx, 2017-12-08:
There are some places where views "zoom" into components even when they shouldn't.

- when opening an existing project, the view is animated until the board/schematic fits into view. This shouldn't be done when the project is opened: there's no existing view to animate "from".
- in the board editor, the footprint view is animated every time the footprint changes. This is quite distracting, and it doesn't make sense. Simply fit to view the new footprint immediately.

ubruhin, 2017-12-11:
>when opening an existing project, the view is animated until the board/schematic fits into view. This shouldn't be done when the project is opened: there's no existing view to animate "from".

Actually I like that animation :smile: In this first half second after opening a project the user is typically not yet ready for any interaction anyway, so IMO it should not really annoy...

> in the board editor, the footprint view is animated every time the footprint changes. This is quite distracting, and it doesn't make sense. Simply fit to view the new footprint immediately.

Here I agree because it's really distracting.

wavexx, 2017-12-11:
On Mon, Dec 11 2017, U. Bruhin wrote:
> Actually I like that animation :smile: In this first half second after
> opening a project the user is typically not yet ready for any
> interaction anyway, so IMO it should not really annoy...

I actually didn't try [I'm away at the moment], but is it possible to
start librepcb on an existing project?

If the last view location is know, it would actually be nice to use that
time to move from fit-to-view to last-location.

ubruhin, 2025-05-24:
Resolved in #1520

closed 2025-05-24 by ubruhin