LibrePCB/LibrePCB #215: Replace references to gerbv

dbrgn, 2017-12-18:
In the docs and in the gerber export dialog you recommend gerbv. However, I've had experiences with gerbv where it would render gerbers on a wrong way (even with examples directly from the reference).

I'd rather recommend Gerbview (part of KiCad, docs [here](http://docs.kicad-pcb.org/stable/en/gerbview.html)) and/or the [online reference viewer](https://gerber.ucamco.com/).

ubruhin, 2017-12-18:
Hmm okay, actually I was happy with gerbv until now (more than with Gerbview). Do you have example files which gerbv doesn't render properly? But yes, the online reference viewer is probably a good candidate to recommend.

marmei, 2018-01-06:
What do you think about kayout? https://github.com/klayoutmatthias/klayout. It is native Qt.

ubruhin, 2018-01-06:
> What do you think about kayout?

I have never heard about it, so I spent a few minutes to give it a try but unfortunately I was not even able to open a Gerber file with it :wink: Did you try it too?

marmei, 2018-01-06:
Yes Indeed, Klayout is a professional tool for chip design. I used it for MEMS design and it is much better suited than all commercial tools I have tried/provided by my university.
However I do not know if it is straight forward to combine it directly with librePCB.

ubruhin, 2018-01-06:
OK but I don't see any reason to "combine" it with LibrePCB, it's something completely different. This issue is just about recommending LibrePCB users a Gerber viewer which they could use to review the generated Gerber files. And until now I never saw a better Gerber viewer than `gerbv`...

marmei, 2018-02-05:
BTW - excellent talk at fosdem! 
In my opinion you are are paving the most promising way to create an open source PCB editor in commercial grade.
Just wanted add this link for viewing gerber files in Klayout. http://klayout.de/klayout_screenshot5_pcb.png

ubruhin, 2018-02-05:
> BTW - excellent talk at fosdem!
In my opinion you are are paving the most promising way to create an open source PCB editor in commercial grade.

Great to hear, thank you! :)

> Just wanted add this link for viewing gerber files in Klayout. http://klayout.de/klayout_screenshot5_pcb.png

Sure that this screenshot shows an imported Gerber file?

But anyway, IMO we should keep the current link to `gerbv` because it is *very* intuitive, already widely used, and also my own experience with it is very positive. It always worked properly for me. @dbrgn told me that the wrong results he saw might be caused by invalid Gerber data, so not really a fatal bug of the viewer (although it would be better to warn about invalid data).

But adding a link to the official online Gerber viewer is still a good idea, we should add a link (in addition to `gerbv` to the Gerber export dialog. **So this issue is now to add such a link to that dialog.**

To also add that link to the documentations, i created https://github.com/LibrePCB/librepcb-doc/issues/2.

dbrgn, 2018-02-06:
Yes, I take back what I said about gerbv :)

kliment, 2018-02-07:
I recently implemented something based on libgerbv and it's a bit confusing but not at all difficult. Do we want a libgerbv-based viewer built right into librepcb medium-term?

ubruhin, 2018-02-07:
> Do we want a libgerbv-based viewer built right into librepcb medium-term?

medium-term or long-term maybe yes, but with very low priority as it's not a big deal to use a standalone Gerber viewer.

dbrgn, 2018-04-05:
Here's an example where gerbv seems to be wrong:

[image]

Reference viewer:

[image]

ubruhin, 2018-05-11:
This was fixed in #255...

closed 2018-05-11 by ubruhin