LibrePCB/LibrePCB #217: I can't type anything in LibrePCB

fdedraco, 2017-12-22:
I run the .appimage file in ubuntu 16.04 and i can't type anything in the app, not even with sudo

ubruhin, 2017-12-22:
Which keyboard layout are you using? And could you post the full terminal output of LibrePCB here?

Does it work when you start LibrePCB this way?
```
QT_XKB_CONFIG_ROOT=/usr/share/X11/xkb ./LibrePCB-Nightly-Linux-x86_64.AppImage
```

fdedraco, 2017-12-23:
EN US, 
```
fathin@fathin-X455LD:~/apps$ sudo ./LibrePCB-Nightly-Linux-x86_64.AppImage -qws
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Qt: Could not determine keyboard configuration data from X server, will use hard-coded keymap configuration.
Qt: Failed to compile a keymap!
Current XKB configuration data search paths are: 
/usr/share/X11/xkb
Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
[  INFO   ] "LibrePCB 0.1.0 (5d6076a)" (:0)
[  INFO   ] "Qt version: 5.9.1 (compiled against 5.9.1)" (:0)
[  INFO   ] "Resources directory: /tmp/.mount_LibrePJxIE5I/opt/share/librepcb" (:0)
[DEBUG-MSG] Started network access manager thread. (:0)
[DEBUG-MSG] Load workspace settings... (:0)
[DEBUG-MSG] Workspace settings successfully loaded! (:0)
[DEBUG-MSG] Load remote workspace library: "1fcb0725-d2e8-4f5d-892c-d1efb44d2963.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "326f091b-b715-44bf-b385-b613cd60d9f3.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "6ccc516c-21b7-4cd5-9cf2-7a04cfa361c6.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "a9ddf0c6-9b1c-4730-b300-01b4f192ad40.lplib" (:0)
[DEBUG-MSG] Load workspace library database... (:0)
[DEBUG-MSG] Workspace library database successfully loaded! (:0)
[DEBUG-MSG] Stopped network access manager thread. (:0)
[DEBUG-MSG] Exit application with code 0 (:0)

fathin@fathin-X455LD:~/apps$ QT_XKB_CONFIG_ROOT=/usr/share/X11/xkb ./LibrePCB-Nightly-Linux-x86_64.AppImage
Qt: Could not determine keyboard configuration data from X server, will use hard-coded keymap configuration.
Qt: Failed to compile a keymap!
Current XKB configuration data search paths are: 
/usr/share/X11/xkb
Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
[  INFO   ] "LibrePCB 0.1.0 (5d6076a)" (:0)
[  INFO   ] "Qt version: 5.9.1 (compiled against 5.9.1)" (:0)
[  INFO   ] "Resources directory: /tmp/.mount_LibrePHa6kGb/opt/share/librepcb" (:0)
[DEBUG-MSG] Started network access manager thread. (:0)
[DEBUG-MSG] Load workspace settings... (:0)
[DEBUG-MSG] Workspace settings successfully loaded! (:0)
[DEBUG-MSG] Load remote workspace library: "1fcb0725-d2e8-4f5d-892c-d1efb44d2963.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "326f091b-b715-44bf-b385-b613cd60d9f3.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "6ccc516c-21b7-4cd5-9cf2-7a04cfa361c6.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "a9ddf0c6-9b1c-4730-b300-01b4f192ad40.lplib" (:0)
[DEBUG-MSG] Load workspace library database... (:0)
[DEBUG-MSG] Workspace library database successfully loaded! (:0)
[ WARNING ]   OpenType support missing for script 18 (:0)
[ WARNING ]   OpenType support missing for script 18 (:0)
[ WARNING ]   OpenType support missing for script 17 (:0)
[ WARNING ]   OpenType support missing for script 17 (:0)
[DEBUG-MSG] Redirect from "https://api.librepcb.org/api/v1/libraries" to "https://raw.githubusercontent.com/LibrePCB/librepcb-api-static/master/api/v1/libraries" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB/librepcb-api-static/master/api/v1/libraries" (:0)
[DEBUG-MSG] Redirect from "https://github.com/LibrePCB-Libraries/LibrePCB_Base.lplib/raw/master/library.png" to "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_Base.lplib/master/library.png" (:0)
[DEBUG-MSG] Redirect from "https://github.com/LibrePCB-Libraries/LibrePCB_Connectors.lplib/raw/master/library.png" to "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_Connectors.lplib/master/library.png" (:0)
[DEBUG-MSG] Redirect from "https://github.com/LibrePCB-Libraries/LibrePCB_ICs.lplib/raw/master/library.png" to "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_ICs.lplib/master/library.png" (:0)
[DEBUG-MSG] Redirect from "https://github.com/LibrePCB-Libraries/Isabellenhuette.lplib/raw/master/library.png" to "https://raw.githubusercontent.com/LibrePCB-Libraries/Isabellenhuette.lplib/master/library.png" (:0)
[DEBUG-MSG] Redirect from "https://github.com/LibrePCB-Libraries/Molex.lplib/raw/master/library.png" to "https://raw.githubusercontent.com/LibrePCB-Libraries/Molex.lplib/master/library.png" (:0)
[DEBUG-MSG] Redirect from "https://github.com/LibrePCB-Libraries/STMicroelectronics.lplib/raw/master/library.png" to "https://raw.githubusercontent.com/LibrePCB-Libraries/STMicroelectronics.lplib/master/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_Base.lplib/master/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_Connectors.lplib/master/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_ICs.lplib/master/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Molex.lplib/master/library.png" (:0)
[... cut ...]
[DEBUG-MSG] project successfully loaded! (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:322)
[DEBUG-MSG] Begin saving the project to temporary files... (:0)
[DEBUG-MSG] Begin saving the project to original files... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] closed project: "/home/fathin/LibrePCB-Workspace/projects/Assembler/Assembler.lpp" (:0)
[DEBUG-MSG] Stopped network access manager thread. (:0)
[DEBUG-MSG] Exit application with code 0 (:0)
```

ubruhin, 2017-12-23:
Why are you passing the argument `-qws`? I guess it's only for Embedded Linux devices...

I tried to reproduce this with the Ubuntu 16.04.3 Live CD in a VirtualBox VM:
1. Choose "English" language and click on "Try Ubuntu"
2. Open the terminal and execute following commands:
    ```
    wget -O LibrePCB-Nightly-Linux-x86_64.AppImage "https://bintray.com/librepcb/LibrePCB-Nightly/download_file?file_path=LibrePCB-Nightly-Linux-x86_64.AppImage"
    chmod +x ./LibrePCB-Nightly-Linux-x86_64.AppImage
    ./LibrePCB-Nightly-Linux-x86_64.AppImage
    ```
3. LibrePCB starts and works as expected (no issues with keyboard)

fdedraco, 2017-12-24:
`-qws` was my previous attempt before trying your suggestion. 

maybe it's just my xkb i'll try to reset it

closed 2017-12-24 by fdedraco

ubruhin, 2017-12-24:
OK, just let us know if there is still something wrong with the AppImage.