LibrePCB/LibrePCB #220: Project "sales pitch"

jupart, 2018-01-04:


ubruhin, 2018-01-04:
You're right, it's still on my ToDo list to create a comparison between LibrePCB and other EDA tools (see also https://github.com/LibrePCB/LibrePCB.github.io/issues/1) :wink:

Btw, I will give an introduction to LibrePCB at FOSDEM, maybe you are interested in that: http://librepcb.org/blog/2018-01-02_librepcb_at_fosdem_18/

jupart, 2018-01-04:
I'll definitely be checking that out (and keeping an eye on this project), thanks!

ubruhin, 2023-02-21:
I think the new website should now give a much better overview about the advantages of LibrePCB, so let's close this issue for now. If there are still requests to add more information, please report it at the repository https://github.com/LibrePCB/librepcb-website.