LibrePCB/LibrePCB #221: Can't place trace on components pad

shakenov-chinga, 2018-01-06:
I checked out the latest commit on master branch.

It shows error window with "No trace,via or pad at cursor position." when I try to place trace. It works good if I place via and append trace.

Tested on a single resistor and capacitor which connected with each other in scheme.

ubruhin, 2018-01-06:
Are the pads SMT or THT? For SMT pads you first need to select the layer in the dropdown list in the command toolbar to the same layer as the pad is (top_cu or bot_cu, depending on whether the pad is on top or on bottom). Automatically changing the layer is not yet implemented... ;)

Maybe this was the problem?

shakenov-chinga, 2018-01-06:
That are SMT pads. Layer is top_cu (red) and it is already selected.
 I've looked through the code, but I couldn't find anything that I can help :)

I'm using Linux Mint, by the way

ubruhin, 2018-01-06:
OK, another idea: There is an issue with the grid. Maybe you need to make the grid interval smaller so the cursor can snap on the pad.

But anyway, the board editor is currently **very** experimental, and there are many missing features and bugs ;)

shakenov-chinga, 2018-01-06:
It works now, thanks.

PS. You are doing great job, guys. Accept my words of appreciation, please.

ubruhin, 2018-01-06:
Great!
> PS. You are doing great job, guys.

Thanks :)

closed 2018-01-06 by ubruhin