LibrePCB/LibrePCB #224: Eagle import: UUID list file sensible defaults

hephaisto, 2018-01-11:
The eagle importer needs a (temporary? - please clarify) UUID list file.
If no such file is given, library creation does seem to succeed, however the referencing UUIDs are generated new for every Symbol/Package/Device step, which causes problems later on.
I created a new empty file for this, and it seemed to work. I am not sure however if this is the intended way to use it (please clarify).
I suggest to do one of the following to make it more clear to users:

- Disable the convert buttons until a valid UUID list file is selected
- use a default filename somewhere safe
- use a default hidden file if none is given

Also, there should be some sort of tooltip which explains what should be selected here (either a specific file or a random new file which can be deleted afterwards).

ubruhin, 2018-01-11:
Actually the eagle importer was not designed for end users, I just developed it for myself to easily create some initial libraries (at the time I wrote it, the library editor did not yet exist so this was the only way to create libraries).

The UUID list is required to make the converted libraries reproducible, so you can run it several times and the result will always be the same. This was important for me to not break projects which already use the converted library entities. The UUID list I have created so far is located here: https://github.com/LibrePCB/demo-workspace/blob/master/v0.1/libraries/local/EagleImport/UUID_List.ini

But I think this concept is not suitable for end users because it's too complex and thus probably annoying. The importer needs to be *highly* simplified to make it really usable. I think we should remove the need for the UUID list completely, even if then the results are no longer reproducible (new UUIDs will be generated every time). This will probably still be fine for most users because typically one only needs to convert a library once.

In addition, the importer is very hacky and buggy, this needs to be fixed too ;)

So there is much work to do before it's really usable...

EternityForest, 2018-12-30:
What if the UUIDs were generated from a hash of the Eagle data that produced it?

Absolute reproducibility, and as a bonus it guarantees any change results in a different UUID.

IMHO library import is pretty important, sites like SnapEDA are so handy!

ubruhin, 2019-01-01:
> As a user I'd really be fine with either way, any Eagle support at all would be a big time saver.

Yes, I agree. I described by idea how it could be integrated here: https://librepcb.discourse.group/t/plan-with-libraries/25/4

closed 2021-11-26 by ubruhin