LibrePCB/LibrePCB #225: Library editor segfaults on table element change

hephaisto, 2018-01-13:
At multiple points in the library editor, segfaults occur after changing checkboxes or dropdown menus.
This particular example was when checking a "required" field when adding a new component. Another example is the pad assignment (auto-assignment works).
Stacktrace: https://pastebin.com/GpWMEShA

ubruhin, 2018-01-14:
Can you describe the exact steps you have done to trigger the segfault? And which Qt version do you use?

I clicked around in the library editor a few minutes but was not able to trigger a segfault...

hephaisto, 2018-01-15:
Here an example with a completely new lib (using the demo workspace):
https://www.xup.in/dl,19886705/librepcb_create_segfault.mp4/
I have Qt 5.7.1 as reported by qtdiag. I use the packages from debian stable.

ubruhin, 2018-01-24:
Strange, I'm not able to reproduce this bug with Qt 5.5.1 I have installed on my computer, and with Qt 5.9.1 which is used in the nightly AppImage.

Did you try to remove the whole output directory before compiling it again? Sometimes I get such strange segfaults because `make` does not rebuild all object files of modified source files (probably there is an issue in resolving `#include` dependencies). A complete rebuild fixes these segfaults...

ubruhin, 2019-05-04:
Now with the official LibrePCB 0.1.1 binaries for Linux I'm able to reproduce this. I suspect it happens only with specific Qt versions (official binaries for Linux now use Qt 5.10.1).

I'm already working on a fix for this...

PR #456 (2019-06-06): Library editor: Refactor all table widgets using Qt's model/view framework
In the library editor there are many table widgets, and most of them have several issues (missing sort functionality, not updated when underlying data changed, buggy, ...). This PR completely refactors all these tables by using [Qt's model/view framework](https://doc.qt.io/qt-5/model-view-programming.html). This way the code is much more clean and less error prone, and features like sorted tables are much easier to implement.

In particular, this change improves following things:
- Fix application crashes with newer Qt  [... cut]
closed 2019-06-06 by ubruhin