LibrePCB/LibrePCB #229: Multiple PCB views

UweBonnes, 2018-02-06:
Hello,

when a late design change requires a new route to be laid out from one edge of a complex board to the opposite edge, two simultanious views of the board would be useful:
- one magnified view of the areas where you actual route
- another global view showing where you are in the context of the whole board.

dbrgn, 2018-02-06:
We just discussed that yesterday 🙂 I added it to the [wishlist](https://github.com/LibrePCB/LibrePCB/wiki/Wishlist).

UweBonnes, 2018-02-06:
Somebody asked me on FOSDEM to enter my wish.

closed 2018-02-06 by UweBonnes

dbrgn, 2018-02-06:
Yes, that was probably @ubruhin. We will reopen this issue once it becomes feasible to implement it, for now the priority is to get a first release out though.