LibrePCB/LibrePCB #237: Ellipses cannot be exported as Gerber

ubruhin, 2018-02-13:
Currently we have two different types for pure graphical objects:
- [Ellipse](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/common/geometry/ellipse.h): For circles (radius x == radius y) and ellipses (radius x != radius y)
- [Polygon](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/common/geometry/polygon.h): For squares, rectangles, arcs, and arbitrary polylines (open) or polygons (closed)

I guess my idea with the Ellipse class was to have a more generic type than just a circle (the same type can be used for circles and ellipses). But now I realize that ellipses are not supported by Gerber, so we cannot export them 😭 

There are many different ways to fix this issue, but I think the most suitable is to just rename `Ellipse` to `Circle` and let it have only one radius instead of two for x/y.

Maybe some time we could also add types for commonly used shapes like squares, rectangles or obrounds. But at the moment we can minimize the effort and complexity by just using one type (`Polygon`) for all these shapes. Except for circles, which is a completely different type of shape (although it *could* also be represented by a `Polygon`, but that would be quite strange).

rnestler, 2018-02-14:
This shouldn't be too hard right? So also a good start for a new contributor?

ubruhin, 2018-02-14:
Yes, shouldn't be too hard to implement. I will add the "easy" label.

Another question would be if a circle should be defined by a radius or a diameter. I think technically it doesn't really matter, but maybe specifying a diameter is more intuitive than a radius?

0xB767B, 2018-02-14:
I would strongly disagree saying there is no technical difference between radius and diameter. 

Theoretically, the definition of a circle is:
"A circle is the set of all points in a plane that are at a given distance from a given point, the centre;" (check out wikipedia for more details). We would for sure call this given distance radius.

Now, practically speaking, diameter does not make a lot of sense. Diameter is almost totally unrelated to the center of the circle. If you want to represent a circle in a data-structure, you most likely will specify its center. If we use diameter, the origin could be any arbitrary point on the circumference of the circle. Unless you divide the diameter by 2, which brings you back to.... The radius.

Just my five cents...

ubruhin, 2018-02-15:
I think we should not decide between radius and diameter from the academical point of view, but from the user point of view. And the user probably doesn't care about  the exact definition of a circle, he's more interested in a definition which is easy to use.

My thoughts are following: In a footprint, circles are often drawn according a datasheet, for example to draw the outline of a hole, for the pin-1 marking on the silkscreen, or for the geometry of the package itself. So they are basically copied from a datasheet. And how are circles/holes specified in datasheets? Probably always by diameter, like this:

[image]

[image]

[image]

[image]

So, why should we specify a circle with its radius, if the whole electronic industry uses diameter?

dbrgn, 2018-02-15:
I think for the user it's irrelevant how the data is stored.

When editing an object, we could actually show two input boxes: One for radius and one for diameter. Changing one also updates the other input field. (Alternatively make a dropdown to change the mode, but I think just showing both radius and diameter would be easier to use.)

ubruhin, 2018-02-20:
> I think for the user it's irrelevant how the data is stored.

I think it's not, otherwise we could use a binary file format ;)

IMHO it would be most intuitive when the diff in a file shows exactly what the user did in the GUI. If he changed the diameter of a circle from 2mm to 3mm, the diff on that file should show that the diameter was changed from 2mm to 3mm. If the user changed the radius from 1mm to 1.5mm, the diff should  show that the radius has changed from 1mm to 1.5mm.

But if diameters are much more often used than radius, I think we can focus only on the diameter to keep it simple and still intuitive for most cases. In the rare cases when the user wants to specify a radius, he either need to calculate by himself, or the UI may provide the ability to optionally insert a radius instead of a diameter (like @dbrgn proposed above).

hephaisto, 2018-03-06:
I think there are a lot of *internal* use-cases where radius is more appropriate, e.g. calculation of clearances etc - so I think the *internal* representation should always be radius.

I would agree with @dbrgn that the best GUI representation would be two separate boxes that are updated when the other one changes.

There are indeed cases where a radius is given and not a diameter (e.g. the cape definition of the beagle bone - it took me about 1min to find it). I find it very cumbersome to use CAD-programs that do not allow me to switch between both input methods easily. If it is possible to switch, it is impossible to ensure value-consistency between GUI and file format without introducing ambiguity in the latter. So I think the consistency-argument is void.

Also, I doubt many users will actually look at the diffs and be startled if the values are different from the ones entered in the GUI, especially if there actually is a second input box which shows exactly this value.

sebgonzo, 2018-05-21:
What about using one text box both for diameter and radius, but for radius just add `R` prefix. E.g.
- `1.5mm` diameter
- `R0.75mm` radius
I think this way it's faster to type in than clicking right textbox or selecting dropdown. It's also less error-prone because it's exactly how diameter and radius are shown on mechanical drawings.

I think that in file format it's also important to distinguish between diameter and radius. I agree with @ubruhin that otherwise we could use binary format. E.g. when hole is specified using radius this probably means a mistake.

ubruhin, 2018-06-11:
If nobody already started with this, I would probably fix this issue soon.

> What about using one text box both for diameter and radius, but for radius just add R prefix.

Sounds reasonable. Or maybe just allow simple mathematical functions, so one could insert `1mm*2` into the diameter field, and librepcb converts this to `2mm`. Of course the information whether it was specified by diameter or radius would be lost with this concept...

PR #279 (2018-06-23): Replace ellipses by circles
Use circles instead of ellipses in footprints because ellipses can't be exported to Gerber. For consistency, same change is made for symbols.

Fixes #237.
closed 2018-06-23 by ubruhin