LibrePCB/LibrePCB #247: Buggy Gerber output

dbrgn, 2018-04-06:
gerbv:

[image]

reference viewer:

[image]

oshpark:

[image]

gerbview (0 character only):

[image]

Here's the output:

```gerber
G04 --- HEADER BEGIN --- *
%TF.GenerationSoftware,LibrePCB,LibrePCB,0.1.0*%
%TF.CreationDate,2018-04-06T01:22:20*%
%TF.ProjectId,gerbertest - default,4fa26089-7cc2-42e7-9238-ecfcf38bbe1d,v1*%
%TF.Part,Single*%
%FSLAX66Y66*%
%MOMM*%
G01*
G74*
G04 --- HEADER END --- *
G04 --- APERTURE LIST BEGIN --- *
%ADD10C,1.35*%
G04 --- APERTURE LIST END --- *
G04 --- BOARD BEGIN --- *
D10*
X48721021Y66920000D02*
G75*
G03*
X46721021Y66920000I-1000000J-750821D01*
G01*
G74*
G03*
X46721021Y58920000I7491334J4000000D01*
G01*
G75*
G03*
X48721021Y58920000I1000000J750821D01*
G01*
G74*
G03*
X48721021Y66920000I7491334J4000000D01*
G01*
G04 --- BOARD END --- *
%TF.MD5,165e92c0a5fc09f49da7b3c7b7935a4f*%
M02*
```

Relevant parts (commented):

- `X48721021Y66920000D02` move to 48721021 / 66920000
- `G75` multi quadrant mode (arcs up to 360°)
- `G03` counterclockwise circular interpolation
- `X46721021Y66920000I-1000000J-750821D01` interpolate to 46.. / 66.. with X offset -1.. and Y offset -0.7..
- `G01` linear interpolation
- `G74` single quadrant mode (arcs up to 90°)
- `G03` counterclockwise circular interpolation
- `X46721021Y58920000I7491334J4000000D01` interpolate to 46.. / 58.. with X offset 0.7.. and Y offset 4..
- `G01` linear interpolation
- `G75` multi quadrant mode (arcs up to 360°)
- `G03` counterclockwise circular interpolation
- `X48721021Y58920000I1000000J750821D01` interpolate to 48.. / 58.. with X offset 1.. and Y offset 0.7..
- `G01` linear interpolation
- `G74` single quadrant mode (arcs up to 90°)
- `G03` counterclockwise circular interpolation
- `X48721021Y66920000I7491334J4000000D01` interpolate to 48.. / 66.. with X offset 0.7.. and Y offset 4..
- `G01` linear interpolation

Honestly I think this looks correct, and both gerbv and gerbview seem to be buggy with regards to the single quadrant mode circular interpolation.

Or could it be some kind of rounding issue?

dbrgn, 2018-04-06:
The arcs from 1 to 2 and 3 to 4 seem to be the problem in gerbv and gerbview.

I'll show the arc from point 1 to 2. In counterclockwise single quadrant mode, an arc (circle segment) needs to be drawn from 1 to 2 in counterclockwise direction, so that the angle is less than 90°.

Point 1/A is at `(46.721021, 66.92)`. Point 2/B is at `(46.721021, 58.92)`.

[image]

The offset from point A to the center of the circle is `(7.491334, 4.0)`. That gives us four possible points, C/D/E/F.

[image]

Here is the result when trying to draw a counterclockwise arc from A to B with the center at C, D, E or F. Only `e` is valid (and looks correct).

[image]

Not sure if it could be a rounding issue, but LibrePCB seems correct.

dbrgn, 2018-04-06:
With this file:

```
%FSLAX66Y66*%
%MOMM*%
G04 --- Apertures --- *
%ADD10C,1*%
G04 --- Use Aperture D10 --- *
D10*
G04 --- CCW Circular Interpolation --- *
G03*
G04 --- Draw arc --- *
X46721021Y66920000D02*
G74*
X46721021Y58920000I7491334J4000000D01*
M02*
```

...and this illustration:

[image]

...and this gerbv screenshot:

[image]

...it's obvious that gerbv chooses the wrong quadrant (c instead of e). Both arcs fulfill the angle <= 90° requirement, but c has much higher deviation than e.

I'm not really sure what the fully correct way to deal with this would be. I believe that from all potential arcs, the one with the least difference in the length of AX and BX (where X is the center point) should be chosen.

I'll ask Ucamco via E-Mail, to make sure that these files are not ambiguous.

dbrgn, 2018-04-07:
Patch submitted: https://sourceforge.net/p/gerbv/patches/69/ Let's hope the project is still maintained, last commit was a year ago.

@ubruhin do we want to change something about the way we generate the Gerber files, or can this issue be closed?

ubruhin, 2018-04-07:
Well done @dbrgn!

> do we want to change something about the way we generate the Gerber files, or can this issue be closed?

I think it makes sense to adjust our Gerber generator to work around the bug in libgerbv. Probably by using multi quadrant mode instead of single quadrant mode, but then we need some logic to avoid issues caused by numeric instability (see Gerber specs for details).

dbrgn, 2019-08-14:
Just ran into this again when generating Gerber output. Aisler also has problems correctly rendering a file with this kind of arcs in it.

We should really stop using single quadrant mode arcs, they're hard to implement right and by using multi quadrant mode it's less likely that manufacturers mess up the result.

dbrgn, 2019-08-14:
Nominating for 1.3. I'll try to fix this. @ubruhin if you can point me to the relevant files / location that would be a big help.

ubruhin, 2019-08-15:
Yep, I agree to fix this soon. But we should take care of the numerical instability of multi quadrant mode, see Gerber specs:

> Numerical Instability in Multi Quadrant (G75) Arcs
In G75 mode small changes in the position of center point, start point and end point can swap
the large arc with the small one, dramatically changing the image.
This most frequently occurs with very small arcs. Start point and end point are close together. If
the end point is slightly moved it can end on top of the start point. Under G75, if the start point of
the arc is equal to the end point, the arc is a full circle of 360°, see 4.10.1. A small change in the
position of the end point has changed the very small arc to a full circle.
G75 rounding must be done carefully. Using high resolution is an obvious prerequisite. See
4.16.
The Gerber writer must also consider that the reader unavoidably has rounding errors. Perfectly
exact numerical calculation cannot be assumed. It is the responsibility of the writer to avoid
unstable arcs.
G74 arcs are always less than 90° and this numerical instability does not exist. G74 is
intrinsically stable. Another option is not to use very small arcs, e.g. by replacing them with
draws - the error is very small and draws are stable.

Btw, the relevant files are `libs/librepcb/common/cam/gerbergenerator.cpp` and `libs/librepcb/project/boards/boardgerberexport.cpp`.

closed 2019-08-28 by ubruhin

RealSnoopyDog, 2024-11-01:
One comment: the two points "A" and "B" from the examples above can NOT be drawn with one "single quadrant" operation because they are NOT located within the same quadrant of the circle. As per old gerber spec, this needs two draw operations. One for each quadrant of the circle. 
From the spec: 
Single quadrant arcs must fit entirely within the quadrant in which they begin. A separate data block is required for each quadrant. A minimum of four data blocks is required to plot a circle.

ubruhin, 2024-11-01:
Without taking a deeper look at this, IIRC single quadrant mode has been marked as deprecated in recent Gerber specs and thus doesn't make sense to use anyway. Also with the current implementation didn't have any issues with arcs now for a long time :)

RealSnoopyDog, 2024-11-01:
Yes, I know. I hate this "guess mode" G74 and yes, it was deprecated. But there are still old files which are using this. 
A gerber generator should not use this anymore. A gerber interpreter needs to deal with it if it wants to support all files out in the wild.
I only wanted to point out that the assumptions for the specific example above how the "single quadrant mode" works are wrong.

PR #504 (2019-08-28): Gerber: Always use multi-quadrant mode
@ubruhin here's a first attempt at fixing this issue.

The spec says:

> Another option is not to use very small arcs, e.g. by replacing them with
draws - the error is very small and draws are stable.

Right now I'm replacing arcs shorter than 0.01mm (point-to-point) with straight lines. 

We probably need to handle some special cases, right? A full circle has start==end with angle=360, right? We should probably still draw arcs if start and end are identical.

I guess we can ignore cases where the two points are closer together than 0.01mm with a lage ang [... cut]