LibrePCB/LibrePCB #257: Install on MacOS

DinBan, 2018-04-25:
##### LibrePCB VERSION
LibrePCB-Nightly-Linux-x86_64.AppImage

##### OS / ENVIRONMENT
macOS High Sierra 10.13.4
##### SUMMARY
-bash: ./LibrePCB-Nightly-Linux-x86_64.AppImage: cannot execute binary file
##### STEPS TO REPRODUCE
install Homebrew - 
/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

install wget - 
brew install wget

trying to install LibrePCB -
wget -O LibrePCB-Nightly-Linux-x86_64.AppImage "https://bintray.com/librepcb/LibrePCB-Nightly/download_file?file_path=LibrePCB-Nightly-Linux-x86_64.AppImage" - OK
chmod +x ./LibrePCB-Nightly-Linux-x86_64.AppImage - OK
./LibrePCB-Nightly-Linux-x86_64.AppImage: cannot execute binary file

log from terminal - 

MBP:~ mbp_5.5$ wget -O LibrePCB-Nightly-Linux-x86_64.AppImage "https://bintray.com/librepcb/LibrePCB-Nightly/download_file?file_path=LibrePCB-Nightly-Linux-x86_64.AppImage"
--2018-04-25 11:36:33--  https://bintray.com/librepcb/LibrePCB-Nightly/download_file?file_path=LibrePCB-Nightly-Linux-x86_64.AppImage
Resolving bintray.com (bintray.com)... 75.126.118.184
Connecting to bintray.com (bintray.com)|75.126.118.184|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://dl.bintray.com/librepcb/LibrePCB-Nightly/LibrePCB-Nightly-Linux-x86_64.AppImage [following]
--2018-04-25 11:36:33--  https://dl.bintray.com/librepcb/LibrePCB-Nightly/LibrePCB-Nightly-Linux-x86_64.AppImage
Resolving dl.bintray.com (dl.bintray.com)... 159.122.18.156
Connecting to dl.bintray.com (dl.bintray.com)|159.122.18.156|:443... connected.
HTTP request sent, awaiting response... 302 
Location: https://akamai.bintray.com/2f/2f23892093b6c204bbf5065de8bd8dcaf338912dbb38b75d3174d339e718d02a?__gda__=exp=1524646113~hmac=3ae0c6e7e4c83e4b98842491f9927ee5498d48ad8f263b7340d363ec6353e090&response-content-disposition=attachment%3Bfilename%3D%22LibrePCB-Nightly-Linux-x86_64.AppImage%22&response-content-type=application%2Fx-www-form-urlencoded&requestInfo=U2FsdGVkX1_F8VJyQEzkKNhXpJAWIcz5vWxiG--QMnmr4xhAcsnb3UpLOGticYEkMeTVw1xgbb4ClChtXqr0T7IltkKWffQQc9iRqmCcHR1xS3oCgBx42AtKALXahpYSoy28h0X2jwFQyGvartbzmQ&response-X-Checksum-Sha1=44125658b74dac9efea6c81bb9f76d962352ac1b&response-X-Checksum-Sha2=2f23892093b6c204bbf5065de8bd8dcaf338912dbb38b75d3174d339e718d02a [following]
--2018-04-25 11:36:33--  https://akamai.bintray.com/2f/2f23892093b6c204bbf5065de8bd8dcaf338912dbb38b75d3174d339e718d02a?__gda__=exp=1524646113~hmac=3ae0c6e7e4c83e4b98842491f9927ee5498d48ad8f263b7340d363ec6353e090&response-content-disposition=attachment%3Bfilename%3D%22LibrePCB-Nightly-Linux-x86_64.AppImage%22&response-content-type=application%2Fx-www-form-urlencoded&requestInfo=U2FsdGVkX1_F8VJyQEzkKNhXpJAWIcz5vWxiG--QMnmr4xhAcsnb3UpLOGticYEkMeTVw1xgbb4ClChtXqr0T7IltkKWffQQc9iRqmCcHR1xS3oCgBx42AtKALXahpYSoy28h0X2jwFQyGvartbzmQ&response-X-Checksum-Sha1=44125658b74dac9efea6c81bb9f76d962352ac1b&response-X-Checksum-Sha2=2f23892093b6c204bbf5065de8bd8dcaf338912dbb38b75d3174d339e718d02a
Resolving akamai.bintray.com (akamai.bintray.com)... 23.64.230.171
Connecting to akamai.bintray.com (akamai.bintray.com)|23.64.230.171|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 29499296 (28M) [application/x-www-form-urlencoded]
Saving to: <<LibrePCB-Nightly-Linux-x86_64.AppImage>>

LibrePCB-Nightly-Li 100%[===================>]  28.13M  2.35MB/s    in 12s     

2018-04-25 11:36:47 (2.36 MB/s) - <<LibrePCB-Nightly-Linux-x86_64.AppImage>> saved [29499296/29499296]

MBP:~ mbp_5.5$ chmod +x ./LibrePCB-Nightly-Linux-x86_64.AppImage
MBP:~ mbp_5.5$ ./LibrePCB-Nightly-Linux-x86_64.AppImage
-bash: ./LibrePCB-Nightly-Linux-x86_64.AppImage: cannot execute binary file
****

##### EXPECTED RESULTS
Install LibrePCB
##### ACTUAL RESULTS
Nothing

dbrgn, 2018-04-25:
The Linux binaries won't run on macOS since the binary format differs.

Nightly builds for macOS are not *yet* available. You can track the progress here:  #250

In the meantime, you could try to follow the [development instructions](https://github.com/LibrePCB/LibrePCB#development) to compile the program yourself.

Let us know if you have any other questions! I'll close this issue in favor of #250.

closed 2018-04-25 by dbrgn

ubruhin, 2018-05-21:
Mac nightly builds are now also available, but not yet tested. See https://github.com/LibrePCB/LibrePCB/issues/250#issuecomment-390583119.

@DinBan If you test these nightly builds on Mac, it would be great to get some feedback if it works :) Please use #250 for responding.