LibrePCB/LibrePCB #259: Failed tests are hard to analyze because of unreadable printing

ubruhin, 2018-05-05:
##### LibrePCB VERSION
59a13da8878346878faa329ffcc12760f90b37b3

##### OS / ENVIRONMENT
All

##### SUMMARY

When running tests (executing the `./tests` binary), the output looks nice for successful tests. But failed tests produce unreadable output because objects are printed as binary blobs.

##### STEPS TO REPRODUCE

Create a simple test which always fails:

```
TEST_F(TestClass, testFunction) {
    QString s1("foo");
    QString s2("bar");
    EXPECT_EQ(s1, s2);
}
```

And execute it:

    ./tests --gtest_filter=TestClass.testFunction

##### EXPECTED RESULTS

All objects (not just QString) should be printed using their textual representation. Most (all?) Qt classes have a stream operator for debug output, and also in LibrePCB there are many `QDebug operator<<(QDebug debug, const T& obj)` overloadings for our own types. See [QDebug](https://doc.qt.io/qt-5/qdebug.html) for details.

Objects which have such an `operator<<(...)` should be printed in test output using this operator overloading, e.g.:

```
[ RUN      ] TestClass.testFunction
../../LibrePCB/tests/test.cpp:51: Failure
      Expected: s1
      Which is: "foo"
To be equal to: s2
      Which is: "bar"
[  FAILED  ] TestClass.testFunction (1 ms)
```

##### ACTUAL RESULTS

The actual string values are not visible, just the binary blob of the QString objects is printed:

```
[ RUN      ] TestClass.testFunction
../../LibrePCB/tests/test.cpp:51: Failure
      Expected: s1
      Which is: { 2-byte objecast <66-00>, 2-byte object <6F-00>, 2-byte object <6F-00> }
To be equal to: s2
      Which is: { 2-byte object <62-00>, 2-byte object <61-00>, 2-byte object <72-00> }
[  FAILED  ] TestClass.testFunction (1 ms)
```

##### HOW TO FIX

The Google Test Framework [allows to use custom value printing functions](https://github.com/google/googletest/blob/master/googletest/docs/AdvancedGuide.md#teaching-google-test-how-to-print-your-values), but this is done for each class separately and thus not very useful. We need a solution which automatically uses the `QDebug operator<<(QDebug debug, const T& obj)` function if the object to print provides one.

hephaisto, 2018-05-06:
Shouldn't it be possible to use a templated PrintTo function for each namespace and use SFINAE to selectively enable them? A bit annoying that it has do be defined for each namespace, though...

ubruhin, 2018-05-06:
Unfortunately I'm not a template expert, but I also think it should be possible something like that...

hephaisto, 2018-05-06:
This snippet should create a `std::ostream& operator<<(std::ostream&, const T&)` for the cases where there exists a `QDebug& operator<<(QDebug&, T);`:
https://gist.github.com/hephaisto/6c1357a86efc302104bd59fdf7e5e03f

ubruhin, 2024-02-15:
Closing because it's not that important and already unresolved for a very long time...

closed 2024-02-15 by ubruhin