LibrePCB/LibrePCB #264: Implement keyboard shortcuts to zoom in Schematic Editor.

Petross404, 2018-05-20:
I like to use smooth scrolling for my mouse, but this causes LibrePCB to zoom in/out way too fast. I can't zoom to a level that I can work with, so I want to try and implement keyboard shortcuts (Ctl +/-) to zoom incrementally in small steps.

I would appreciate if someone could help to pinpoint where the source code for [Schematic Editor](https://i.imgur.com/La2HtWU.png) lives.

EDIT : I think it's inside `libs/librepcb/schematiceditor/schematiceditor.cpp`

ubruhin, 2018-05-20:
> I like to use smooth scrolling for my mouse

How do you mean that, did you make some special configuration for that?

> but this causes LibrePCB to zoom in/out way too fast.

Does other software work properly with your setup? For example CADs like LibreCAD, FreeCAD, Eagle, ...

> so I want to try and implement keyboard shortcuts (Ctl +/-) to zoom incrementally in small steps.

Makes sense, but I think it's more important to get zooming by mouse wheel working properly. Even with smooth scrolling it should work as expected.

> I would appreciate if someone could help to pinpoint where the source code for Schematic Editor lives.
> EDIT : I think it's inside libs/librepcb/schematiceditor/schematiceditor.cpp

Right. But the code for zooming stuff is in `libs/librepcb/common/graphics/graphicsview.cpp`.

Petross404, 2018-05-22:
Is the following code ok? Do you want to format it in another way?
```cpp
void SchematicEditor::keyPressEvent(QKeyEvent* k_event)
{
	if (k_event->key() == Qt::Key_Plus)
	{
		mGraphicsView->zoomIn();
	}
	else if (k_event->key() == Qt::Key_Minus)
	{
		mGraphicsView->zoomOut();
	}
} 
```
Of course I will create a PR, I am just asking...

ubruhin, 2018-05-22:
> EDIT : This code only works when the grid is selected and active. Do you want a keyboard shortcut for the actions so it can work when SchematicEditor window is active?

Ah, do you mean to just set the keyboard shortcuts to the "zoom in" and "zoom out" actions which are already available in the toolbar? That would actually be the cleanest solution IMHO (can even be done in the designer, no code needed for that). Did you try if it works?

Anyway, it would be great to have these shortcuts in all three editors (library/schematic/board).

Petross404, 2018-05-22:
> Ah, do you mean to just set the keyboard shortcuts to the "zoom in" and "zoom out" actions which are already available in the toolbar?

Yes. If you are OK with I can use the Designer allthough my personal preference is to make it happen inside the code. Of course it's easier for me too, to use the Designer.

> all three editors (library/schematic/board)

That's about the window I posted plus 2 other?

ubruhin, 2018-05-22:
> If you are OK with I can use the Designer allthough my personal preference is to make it happen inside the code.

Yes that would be fine (at least then it's consistent, other actions also have their shortcut set in the designer).

> That's about the window I posted plus 2 other?

Correct. You can find them with Ctrl+K in qtcreator and type "boardeditor" resp. "libraryeditor".

Petross404, 2018-05-23:
I hope [this](https://github.com/LibrePCB/LibrePCB/pull/266) one is sufficient.

ubruhin, 2018-05-24:
Fixed in #266. Thanks @Petross404!

closed 2018-05-24 by ubruhin

Petross404, 2018-05-24:
It was my pleasure, I hope I can help again in the future. Have fun with coding and thanks for your work guys!