LibrePCB/LibrePCB #270: Set up community translation project

dbrgn, 2018-05-27:
We should have a simple way for the community to contribute and fix translations.

Transifex is a great tool for this, and free for open source: https://www.transifex.com/

TODO:

- [ ] Set up Transifex for a test run
- [ ] Document how to contribute translations

ubruhin, 2018-05-28:
Yep, good idea :+1: 

I created a project here: https://www.transifex.com/librepcb/librepcb-application/.

dbrgn, 2018-05-29:
Nice!

We should also create a translation guide. One part of that is "how to translate", the other one should be how to consistently translate technical terms.

We should for example create a table for all languages where we can specify how to translate words like "component" or "device". Otherwise we'll end up with inconsistent and confusing translations.

ubruhin, 2019-06-06:
@dbrgn I guess translation works pretty good so far, so can we close this issue?

dbrgn, 2019-06-06:
Yes!

closed 2019-06-06 by dbrgn