LibrePCB/LibrePCB #274: Crash (SIGABRT) on macOS

dbrgn, 2018-06-05:
##### LibrePCB VERSION

Current nightly (via https://download.librepcb.org/nightly_builds/master/librepcb-installer-nightly-mac-x86_64.dmg)

##### OS / ENVIRONMENT

Mac OS X 10.13.4 (17E199)

##### SUMMARY

A friend tested whether the current nightly build works on his mac. The setup went fine and installed LibrePCB, but when starting LibrePCB it crashes reproducibely:

```
Process:               librepcb [17314]
Path:                  /Users/USER/*/librepcb.app/Contents/MacOS/librepcb
Identifier:            test.librepcb
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           librepcb [17314]
User ID:               501

Date/Time:             2018-06-05 09:28:28.319 +0200
OS Version:            Mac OS X 10.13.4 (17E199)
Report Version:        12
Anonymous UUID:        XXX

Sleep/Wake UUID:       XXX

Time Awake Since Boot: 2300000 seconds
Time Since Wake:       85000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6facdb6e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff6fc98080 pthread_kill + 333
2   libsystem_c.dylib             	0x00007fff6fa291ae abort + 127
3   test.librepcb                 	0x00000001014d3417 0x101277000 + 2475031
4   org.qt-project.QtCore         	0x00000001022f4b6f 0x1022d8000 + 117615
5   org.qt-project.QtCore         	0x00000001022eedba 0x1022d8000 + 93626
6   org.qt-project.QtCore         	0x00000001022f07ee QMessageLogger::fatal(char const*, ...) const + 148
7   test.librepcb                 	0x00000001014c133d 0x101277000 + 2401085
8   test.librepcb                 	0x00000001014c114e 0x101277000 + 2400590
9   test.librepcb                 	0x00000001012893a1 0x101277000 + 74657
10  libdyld.dylib                 	0x00007fff6f97d015 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff6face292 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff6fc9520e _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff6fc94be9 start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff6face292 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff6fc95009 _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff6fc94be9 start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff6face292 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff6fc95009 _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff6fc94be9 start_wqthread + 13
[... 281 lines cut ...]
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Accelerate framework               128K        2 
Activity Tracing                   256K        2 
CG backing stores                  440K        3 
CoreImage                            8K        3 
CoreServices                        72K        2 
CoreUI image file                  280K        3 
Kernel Alloc Once                    8K        2 
MALLOC                            97.2M       25 
MALLOC guard page                   32K        7 
STACK GUARD                       56.0M        8 
Stack                             11.0M        8 
VM_ALLOCATE                         56K        6 
__DATA                            22.8M      232 
__FONT_DATA                          4K        2 
__LINKEDIT                       196.7M       14 
__TEXT                           154.2M      235 
__UNICODE                          560K        2 
mapped file                       51.7M       15 
shared memory                     2676K       13 
===========                     =======  ======= 
TOTAL                            594.2M      565 

```

dbrgn, 2018-06-05:
@schmijos do you have the same OS version?

ubruhin, 2018-06-05:
When running librepcb from console, is there any output visible?

crinq, 2018-06-07:
Console output:
```
./generated/mac/librepcb.app/Contents/MacOS/librepcb
[EXCEPTION] The directory "/Users/crinq/sandbox/LibrePCB/build/generated/mac/librepcb.app/Contents/share/librepcb/fontobene" does not exist. (../../../../libs/librepcb/common/fileio/fileutils.cpp:161)
[CRITICAL ] Failed to load stroke font pool: "The directory \"/Users/crinq/sandbox/LibrePCB/build/generated/mac/librepcb.app/Contents/share/librepcb/fontobene\" does not exist." (:0)
[EXCEPTION] The font "newstroke.bene" does not exist in the font pool. (../../../../libs/librepcb/common/font/strokefontpool.cpp:65)
[  FATAL  ] Default stroke font could not be loaded! (:0)
Abort trap: 6
```

crinq, 2018-06-07:
fix:
```
mkdir generated/mac/librepcb.app/Contents/share
mkdir generated/mac/librepcb.app/Contents/share/librepcb
cp -r  ../share/librepcb/fontobene generated/mac/librepcb.app/Contents/share/librepcb/
```

rnestler, 2018-06-07:
So the exception leads to the "Crash (SIGABRT)" mentioned by @dbrgn? Is this expected behavior?

ubruhin, 2018-06-07:
Thanks @crinq for the details.

So the problem is that the "share" directory is not bundled with the application. That directory needs to be located at the path `../share` relative to the application executable. But on mac the application is a directory (*.app) instead of a single executable, which breaks the relative path to the share directory.

> So the exception leads to the "Crash (SIGABRT)" mentioned by @dbrgn?

No, it's not the exception, it's a `qFatal()` statement which calls `abort()`.

> Is this expected behavior?

Yes, if runtime resources (i.e. the "share" directory) are not available when starting librepcb, it aborts with `qFatal()`.

dbrgn, 2018-06-07:
> Yes, if runtime resources (i.e. the "share" directory) are not available when starting librepcb, it aborts with qFatal().

What about an error message dialog, followed by exiting with an error code?

ubruhin, 2018-06-07:
> What about an error message dialog, followed by exiting with an error code?

I'm not sure if this would work properly. The error occurs very early in the startup process, it's not sure if GUI things already work at that point. Trying to open GUIs at this point may lead to even worse errors. 

As this is a really fatal error caused by wrong packaging, I don't think it's a real issue if there is no graphical message box... The user couldn't do anything to get librepcb running anyway, even if he would see an error message - the release is just completely broken.

schmijos, 2018-06-11:
Ups sorry, I apparently have opened the wrong app in my tests.

ubruhin, 2018-06-16:
I tried to fix this in #278.

Could maybe someone try if following nightly build works properly? https://download.librepcb.org/nightly_builds/274-fix-macos-deployment/librepcb-nightly-mac-x86_64.dmg

schmijos, 2018-06-19:
I tried it (this time for real). It works for me.

ubruhin, 2018-06-19:
> I tried it (this time for real). It works for me.

Cool, thank you! :)

PR #278 (2018-06-23): Fix macOS deployment issue causing a crash
Adds the "share" directory to the macOS application bundle to make runtime resources available.

Hopefully fixes #274.
closed 2018-06-23 by ubruhin