LibrePCB/LibrePCB #283: librepcb.desktop: error: value "0.1" for key "Version" in group "Desktop Entry" is not a known version

Petross404, 2018-06-26:
As mentioned [here](https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-1.0.html#recognized-keys) entry `Version`, accepts numeric values from 1 - 3. These values have nothing to do with the application's version but rather with the 

> Version of the Desktop Entry Specification that the desktop entry conforms with

If you have trouble deciding which value is the right one, you can omit this entry alltogether. :)

ktrace, 2018-06-26:
"Version of the Desktop Entry Specification that the desktop entry conforms with. Entries that confirm with this version of the specification should use 1.1. Note that the version field is not required to be present."
https://specifications.freedesktop.org/desktop-entry-spec/latest/ar01s06.html

ubruhin, 2018-06-26:
This was already fixed a long time ago in #176.

I see that you use the "first_pcb" tag in the gentoo package, that's actually not a good idea because this tag is very very old. You should better use the current master branch, i.e. 68577ecf8f39299ef4d81ff964b01c3908d1f10b.

Petross404, 2018-06-27:
> I see that you use the "first_pcb" tag in the gentoo package, that's actually not a good idea because this tag is very very old. You should better use the current master branch, i.e. 68577ec.

I know, but gentoo devs insist on not adding a live (ie current master) ebuild alone. So until a more recent release (even if it's beta) happens, I must use this tag too.

I am closing this myself as it isn't usefull any more. :)

ubruhin, 2018-06-27:
> I know, but gentoo devs insist on not adding a live (ie current master) ebuild alone.

Even when using a specific commit hash (instead of `master`)?

Petross404, 2018-07-26:
> Even when using a specific commit hash (instead of master)?

This ebuild should have a version, I don't know how I am supposed to version this commit.

ubruhin, 2018-07-26:
For our installers (which are updated automatically with every commit) we use the version number "0.1.0-COMMITCOUNT", while COMMITCOUNT is the number of commits on `master`. It's the same number as shown on GitHub:

[image]

So the current version number is "0.1.0-1274". Maybe you could use the same concept for the Gentoo package?

Petross404, 2018-07-28:
I like that. I will ask the devs.

rnestler, 2018-09-18:
With the first release getting closer (https://github.com/LibrePCB/LibrePCB/milestone/1) this issue will probably be solved.

ubruhin, 2018-11-25:
@Petross404 As there is now an official stable release [available](https://librepcb.org/blog/2018-11-25_release_0.1.0/), I guess this issue can be closed. Feel free to reopen if you disagree :)

closed 2018-11-25 by ubruhin

Petross404, 2018-11-26:
Excellent news!