LibrePCB/LibrePCB #286: Style guide: #include sorting

dbrgn, 2018-07-08:
This is about the style guide.

Right now there does not seem to be any fixed ordering/sorting of includes. Furthermore, sometimes relative includes are used (`#include "../dialogs/projectpropertieseditordialog.h"`) and sometimes not (`#include <librepcb/common/dialogs/boarddesignrulesdialog.h>`).

Sorting of includes has the advantage that pull requests adding includes conflict less often. Is there a de-facto standard for include ordering? Maybe something that could be included in the QtCreator style file?

Otherwise I'd suggest the following ordering:

- 1: C++ imports (`#include <iostream>`)
- 2: Qt imports (`#include <QtCore>`)
- 3: Imports from other modules, always starting with `librepcb` (`#include <librepcb/project/project.h>`)
- 4: Imports relative to the current module (`#include "fsm/bes_fsm.h"`)

All imports should be sorted alphanumerically inside their group.

@ubruhin @rnestler

ubruhin, 2018-07-08:
Actually I would like to use a code formatting tool like `clang-format`, which would also reorder include statements. But then I think we should also check on CI if the code was formatted properly (to avoid commits which format code just because someone forgot to run the formatter in an older commit).

I just wonder if this would frustrate contributors because either they need to install and run a code formatting tool, or CI on pull requests will fail because of wrong formatting...

dbrgn, 2018-07-09:
I think it would be OK if the formatter were the first step in CI, so that you don't have to wait an hour just to see that your code isn't formatted correctly :)

rnestler, 2018-07-09:
> I just wonder if this would frustrate contributors because either they need to install and run a code formatting tool, or CI on pull requests will fail because of wrong formatting...

I think if it is documented in the README or CONTRIBUTING file and there are instructions on how to install and run clang-format, people won't be frustrated.

dbrgn, 2018-07-10:
At least on Arch, clang-format is shipped with clang.

The only problem I can imagine is if people use gcc and don't want to install clang... But in that case we could also add instructions on how to run clang-format through a Docker container?

ubruhin, 2018-07-10:
@dbrgn You know that there are even operating systems without a package manager? :wink:

dbrgn, 2018-07-10:
Yes of course. But I would guess that the majority of *development* (not necessarily deployment) of LibrePCB will happen on systems with a proper package manager :wink:

ubruhin, 2018-09-05:
Another option would be to check in a `.clang-format` file, run clang-format once and commit all the changes. Then contributors may or may not run clang-format by themselves (but only on those files they changed!), and CI does not check for style violations. From time to time (maybe 1-4x per year) we could just add a commit to re-format all the code again.

Maybe that would be a good compromise; The code would be formatted very well, and contributors are not frustrated because CI reports style violations.

ubruhin, 2018-09-06:
> It has the downside of obscuring git blame workflows with reformatting.

Thus I would reformat it only very rarely...

ubruhin, 2018-09-06:
> But then you'll have improperly formatted code in the codebase for most of the time

We have improperly formatted code since over 5 years ;) Formatting it *once* would not be perfect, but at least *much* better than the current situation.

Tubbles, 2018-09-06:
How about installing a server-side git hook which refuses pushes if they are not formatted correctly? I think it is possible

dbrgn, 2018-09-06:
> How about installing a server-side git hook which refuses pushes if they are not formatted correctly?

We could do something like that easily by adding a formatting step to CI. The main question is whether that's too much of a burden for developers, as far as I can tell, right?

> Formatting it once would not be perfect, but at least much better than the current situation.

I fully agree. But I think reformatting it once now and ensuring that future commits are properly formatted would be even better :sunglasses:

ubruhin, 2018-09-06:
One big problem with clang-format is that its output depends on the used version. So it only works properly if *everyone* uses the exact same version which is also used on CI. But that's pretty annoying, for example on Linux it depends on the used distribution what version for clang-format you get.