LibrePCB/LibrePCB #288: Improve error messages for invalid library elements

dbrgn, 2018-07-17:
##### LibrePCB VERSION

Commit d7d9096ec

##### OS / ENVIRONMENT

Arch Linux

##### SUMMARY

Please tell me what I did wrong :slightly_smiling_face: 

[image]

dbrgn, 2018-07-17:
It seems that the reason in this case was duplicate pin names.

1. Why is this disallowed? The pins have a UUID for disambiguation.
2. There should be feedback already when creating the invalid pin. We could colorize all duplicate pin names with red.
3. In the error message, there should be a hint on what the user did wrong.

dbrgn, 2018-07-17:
> if they are electrically connected, one GND pin in schematics is enough

That's one of the points where I was very unsure on how to model the symbol. It would be really useful if that could be shown somewhere. Similar to the package wizard, where you show this useful hint:

[image]

ubruhin, 2018-08-02:
> It would be really useful if that could be shown somewhere. Similar to the package wizard, where you show this useful hint:

I did not yet integrate it into the symbol editor (in contrast to the package wizard, the symbol wizard has no page to add pins), but at least I added a note to the getting started guide: https://docs.librepcb.org/#gettingstarted-libraries-sym

In addition, with #296 it will be possible to save a symbol even if it has multiple pins with the same name. Later we can show at least a warning in that situation...

closed 2018-08-03 by ubruhin