LibrePCB/LibrePCB #289: crashes on startup after fresh installation

exocode, 2018-07-19:
##### LibrePCB VERSION
0.1.0-119nightly

##### OS / ENVIRONMENT
Mac High Sierra 10.13.6 (17G65)
##### SUMMARY
Downloaded from docs page, Installed it, tried to run

##### STEPS TO REPRODUCE
see summary

##### EXPECTED RESULTS
app should start

##### ACTUAL RESULTS
crashes

Process:               librepcb [41450]
Path:                  /Users/USER/*/librepcb.app/Contents/MacOS/librepcb
Identifier:            test.librepcb
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           librepcb [41450]
User ID:               501

Date/Time:             2018-07-19 16:04:58.380 +0200
OS Version:            Mac OS X 10.13.6 (17G65)
Report Version:        12
Anonymous UUID:        AD24FFCF-27E0-1392-8009-B4C5BFBE9466

Sleep/Wake UUID:       354F6619-06E6-4EEA-9E96-D2FFE611BE65

Time Awake Since Boot: 57000 seconds
Time Since Wake:       3300 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff5be56b66 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff5c021080 pthread_kill + 333
2   libsystem_c.dylib             	0x00007fff5bdb21ae abort + 127
3   test.librepcb                 	0x00000001052e4417 0x105088000 + 2475031
4   org.qt-project.QtCore         	0x0000000106102b6f 0x1060e6000 + 117615
5   org.qt-project.QtCore         	0x00000001060fcdba 0x1060e6000 + 93626
6   org.qt-project.QtCore         	0x00000001060fe7ee QMessageLogger::fatal(char const*, ...) const + 148
7   test.librepcb                 	0x00000001052d233d 0x105088000 + 2401085
8   test.librepcb                 	0x00000001052d214e 0x105088000 + 2400590
9   test.librepcb                 	0x000000010509a3a1 0x105088000 + 74657
10  libdyld.dylib                 	0x00007fff5bd06015 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff5be5728a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff5c01e009 _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff5c01dbe9 start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff5be5728a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff5c01e009 _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff5c01dbe9 start_wqthread + 13

[... 332 lines cut ...]
Bluetooth: Version 6.0.7f10, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: SanDisk Ext SSD
USB Device: Expansion
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Hub
USB Device: AG06/AG03
USB Device: Gaming Mouse
USB Device: USB2.0 Hub
USB Device: Keyboard Hub
USB Device: Apple Keyboard
USB Device: Reloop KeyFadr
USB Device: MIDIMATE II
USB Device: USB2.0 Hub
USB Device: eLicenser
USB Device: TripleHead2Go SE
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

exocode, 2018-07-19:
Found in a comment:

```

mkdir THE_FOLDER_WHERE_YOU_INSTALLED_LIBREPCB/nightly/bin/librepcb.app/Contents/share
mkdir THE_FOLDER_WHERE_YOU_INSTALLED_LIBREPCB/nightly/bin/librepcb.app/Contents/share/librepcb
cp -r  THE_FOLDER_WHERE_YOU_INSTALLED_LIBREPCB/nightly/share/librepcb/fontobene THE_FOLDER_WHERE_YOU_INSTALLED_LIBREPCB/nightly/bin/librepcb.app/Contents/share/librepcb
```

ubruhin, 2018-07-19:
Ah, it seems that #278 fixed only the portable package, but not the installer package :(

I will fix this soon.

ubruhin, 2018-07-19:
Lol, the root cause of this issue was very trivial:

The version number of installer packages contains the commit count of the `master` branch to have a continuously increasing version number. But Travis-CI clones the repository with a depth of only 50 commits, so the version number has never incremented and thus the installer packages were never updated. So the packages were just outdated and didn't contain the bugfix of #278 :sob: 

I fixed this in a79adb6aee46f9d40bb483ca313abdf126cfe8d3 by forcing Travis to clone the whole history. Now the installer packages are up-to-date and thus should (hopefully) work properly.

@exocode Please just tell me if there is still an issue with the macOS installer.

closed 2018-07-19 by ubruhin