LibrePCB/LibrePCB #293: Multiple projects opened when opening recent project

dbrgn, 2018-07-24:
##### LibrePCB VERSION

Version: 0.1.0 (first_pcb-605-g9c37c48-dirty)
Build date: 2018-07-23 00:19:13 (CEST)

##### STEPS TO REPRODUCE

Here are my recent projects:

[image]

If I doubleclick on the top one, everything works correctly:

[image]

But if I doubleclick on the bottom one, both projects are opened:

[image]

ubruhin, 2018-07-25:
> If I doubleclick on the top one

Maybe here's the problem - only a single click is required, not a double click ;) If you click on the bottom one, the entries are swapped immediately (the most recent one is always at top), so the second click goes to the other project xD

dbrgn, 2018-07-25:
Hm, first of all it takes about 1s after the click without visual feedback until something happens... Furthermore, the swap does not happen until after that 1s, so I definitely doubleclicked on the same entry :)

Do you maybe block the UI thread with something?

ubruhin, 2018-07-25:
> Hm, first of all it takes about 1s after the click without visual feedback until something happens... 

> Do you maybe block the UI thread with something?

This is the time needed for loading the project.

> Furthermore, the swap does not happen until after that 1s

Because the thread is blocked until the project is loaded.

> so I definitely doubleclicked on a single entry 

Both click events are added to the event queue. But because processing of the first click event takes some time, the second click is delayed.

This is how most GUI applications work ;)

dbrgn, 2018-07-25:
> This is how most GUI applications work ;)

Yes, I know how the event queue is being handled, but the thread should not block for an operation that takes that long :slightly_smiling_face: In the end, even as a software developer, I did not find out why the behavior was "strange".

What could help is some kind of "pressing" animation on the element, so that you get immediate feedback that your input was accepted for processing.

ubruhin, 2018-07-26:
> thread should not block for an operation that takes that long

1s is very short for an EDA application :grin: Other tools are sometimes blocking for >>10s...

Of course it would be nice to load the project in another thread, but I think at this time it's not worth the effort.

> What could help is some kind of "pressing" animation on the element, so that you get immediate feedback that your input was accepted for processing.

Maybe we could immediately disable (gray out) the element right after the click and re-enable it after the project was loaded. This would give immediate feedback, but when doing a doubleclick it still opens both projects.

doom-fr, 2019-12-03:
Why just setting the opening to double click ?
It will solve the problem easily.

doom-fr, 2020-04-09:
In fact, a double click is done, not 2 simple clicks.
So maybe adding a double click listener and redirect it to the simple click routine will solve the problem.
It will maybe eat the double click and not consider it as two simple clicks.
It's an idea. Do not know if it works.

PR #958 (2022-03-11): Control panel: Fix UX flaws due to blocking UI when opening project
- Fix opening 2 projects when double-clicking on a recent project (see #293). This is done by delaying the update of `QListView` by 500ms (a bit ugly, but fixes the problem and doesn't really matter since during this delayed update, the ControlPanel is in background anyway since a project was just opened).
- Set "wait cursor" while opening project - when clicking on a project, this gives immediate feedback that the application is busy with loading the project. The cursor is reset to normal once the project is fully opened.

Fi [... cut]
closed 2022-03-11 by ubruhin