LibrePCB/LibrePCB #302: Library editor: can't add symbol to component

SaculRennorb, 2018-08-19:
##### LibrePCB VERSION
0.1.0 (first_pcb-649-g31550f4-dirty)

##### OS / ENVIRONMENT
win7 x64

##### SUMMARY
what is says on the tin: I cant add a symbol in the component creation dialog 

##### STEPS TO REPRODUCE
just following the example in the documentation on how to create local libraries, just with a transistor instead of an ic, after creating a symbol and working my way through the component dialog i cant add a symbol and thus not finish component creation

##### EXPECTED RESULTS
adding a symbol to my custom component in my lib

##### ACTUAL RESULTS
the symbol select dialog opens, and I am able to select a symbol, but when i double click on it or press ok, the select dialog just closes and the uid / name is not added in the comp. creation dialog

ubruhin, 2018-08-19:
Thanks for the bug report! I fixed it, so it should work now (nightly builds will be updated soon).

closed 2018-08-19 by ubruhin

SaculRennorb, 2018-08-19:
nice, you fixed another bug i noticed as well in this commit, was thinking about whether or not not listing symbols without category was a feature or a _feature_ ^^

ubruhin, 2018-08-19:
> nice, you fixed another bug i noticed as well in this commit, was thinking about whether or not not listing symbols without category was a feature or a feature ^^

:grin: Listing symbols without category is a feature, but unfortunately I broke it in #296, together with the bug you reported :wink:

commit 6b080415 (2018-08-19): LibraryEditor: Fix broken SymbolChooserDialog and PackageChooserDialog

- Symbol for Component could not be choosen
- Symbols/Packages without categories were not shown

Fixes #302