LibrePCB/LibrePCB #304: Library editor: unreadable text on create new component dialog (component signals)

SaculRennorb, 2018-08-19:
##### LibrePCB VERSION
 0.1.0 (first_pcb-655-g6b08041-dirty)

##### OS / ENVIRONMENT
win7 x64

##### SUMMARY
when creating a new component via the dialog the leftmost text in the component signals table is unreadable because the font-size is too large / the row height is too small

##### STEPS TO REPRODUCE
create a new component via the new component dialog in the library editor

##### EXPECTED RESULTS
being able to read the text

##### ACTUAL RESULTS
cant read the text

see [Image](https://prnt.sc/kki2al)

SaculRennorb, 2018-08-19:
same problem in the 'overview tab' once the component is finished, see [Image](http://prntscr.com/kki3u7)

ubruhin, 2018-08-19:
Hmm maybe that only occurs on Windows, at least on Linux Mint I can't reproduce it...

PR #319 (2018-08-31): Fix GUI issues on Windows
- Component Editor: Fix unreadable text in signals list
  - Increasing the minimum row width to 17 pixels seems to fix the unreadable text on Windows.
  - Fixes #304 
- AddComponentDialog: Fix size glitches on Windows
  - Reorganize layout: Replace some HBox/VBoxLayouts by GridLayout.
  - Always show (empty) device/package, even if no device selected.
  - Fixes #307
closed 2018-08-31 by ubruhin