LibrePCB/LibrePCB #305: right-clicking on a project in the control panel crashes the application

SaculRennorb, 2018-08-20:
##### LibrePCB VERSION
 0.1.0 (first_pcb-655-g6b08041-dirty)
##### OS / ENVIRONMENT
win7x64

##### SUMMARY
well....; right-clicking on a project in the control panel crashes the application

##### STEPS TO REPRODUCE
right click on a project in the control panel 

##### EXPECTED RESULTS
don't crash the app

##### ACTUAL RESULTS
app crash

```
Problemsignatur:
  Problemereignisname:	APPCRASH
  Anwendungsname:	librepcb.exe
  Anwendungsversion:	0.0.0.0
  Anwendungszeitstempel:	5b79b295
  Fehlermodulname:	ntdll.dll
  Fehlermodulversion:	6.1.7601.24214
  Fehlermodulzeitstempel:	5b626f7c
  Ausnahmecode:	c00000fd
  Ausnahmeoffset:	0002df96
  Betriebsystemversion:	6.1.7601.2.1.0.256.48
  Gebietsschema-ID:	1031
  Zusatzinformation 1:	0176
  Zusatzinformation 2:	0176d25795137e20a797fc4a456d7383
  Zusatzinformation 3:	5460
  Zusatzinformation 4:	546031bc7709a5de96f309e815a56b60
```

sorry, the trace is in german, but i guess the fields are self explanatory by their values

SaculRennorb, 2018-08-20:
visual studio in attached debug mode tells me its a stack overflow:
`bei 0x773ADF96 (ntdll.dll) in librepcb.exe: 0xC00000FD: Stack overflow (Parameter: 0x00000001, 0x00092FFC)`

and cmd as well as powershell don't log a single line

ubruhin, 2018-08-20:
Hmm I cannot reproduce this on Linux... Where do you click exactly (maybe screenshot)?

SaculRennorb, 2018-08-24:
[image]

here is the screenshot

PR #317 (2018-08-30): Fix broken FilePath::isRoot() on Windows, possibly causing application crash
### Issue

`FilePath::isRoot()` did not work on Windows (it always returned `false`, even for a root path like "C:/"). The root cause of this is that `FilePath` strips trailing slashes, so "C:/" is converted to "C:", and it seems that `QDir::isRoot()` does not work without the trailing slash. This could even lead to an application crash (see #305).

### Changes

- Make `FilePath` appending a slash on Windows drive paths (i.e. represent drives by "X:/" instead of "X:" where X is the drive letter). This automatically fi [... cut]
closed 2018-08-30 by ubruhin