LibrePCB/LibrePCB #306: schematic editor: elements can't be removed from schematic via context menu

SaculRennorb, 2018-08-20:
##### LibrePCB VERSION
 0.1.0 (first_pcb-655-g6b08041-dirty)
##### OS / ENVIRONMENT
win7x64

##### SUMMARY
the context menu option `remove component xxx` doesn't work, selecting the component and pressing the delete key *does work* tho

##### STEPS TO REPRODUCE
use the context menu to delete a component from a schematic

##### EXPECTED RESULTS
component should be deleted

##### ACTUAL RESULTS
component is not deleted

SaculRennorb, 2018-08-20:
also the mirror option does ... nothing ?

dbrgn, 2018-08-20:
> also the mirror option does ... nothing ?

That's not yet implemented, see #268.

dbrgn, 2018-08-20:
> the context menu option `remove component xxx` doesn't work, selecting the component and pressing the delete key *does work* tho

I can confirm that (also under Linux). Should be an easy fix.

SaculRennorb, 2018-08-20:
ok well i didn't look very long for this one tbh, its another thing anyway and this issue is about the delete option int the context menu

ubruhin, 2018-08-21:
The whole context menu is just not implemented yet... Maybe I'll fix this by myself because I'm currently also working on the context menu of the board editor.

0xB767B, 2018-08-21:
Oh, good to know. I was just about getting into context menues ;-)