LibrePCB/LibrePCB #312: Can't ....... save the project... ?

SaculRennorb, 2018-08-21:
##### LibrePCB VERSION
0.1.0 (first_pcb-655-g6b08041-dirty)

##### OS / ENVIRONMENT
win7x64

##### SUMMARY
here is the error message

---------------------------
Error while saving the project
---------------------------
The project could not be saved!

Error Message:
Could not copy library element "C:\Users\LucasB\AppData\Local\Temp\librepcb\1534784591403_20215\af768fc3-a088-44b4-add9-1308dbf7fa42" to "F:\LibrePCB\Workspace\projects\powampv2\library\cmp\af768fc3-a088-44b4-add9-1308dbf7fa42" because the directory exists already.
Could not copy library element "C:\Users\LucasB\AppData\Local\Temp\librepcb\1534784591490_5601\a9a1e040-acc0-4ef0-9d42-850eb926b4cb" to "F:\LibrePCB\Workspace\projects\powampv2\library\sym\a9a1e040-acc0-4ef0-9d42-850eb926b4cb" because the directory exists already.
---------------------------
OK   
---------------------------

##### STEPS TO REPRODUCE
actually i don't know how the error occurred, t just happend some time ago

##### EXPECTED RESULTS
would like to save

##### ACTUAL RESULTS
cant save

and it also didn't create automatic backups ... loads of work was lost that day :/

ubruhin, 2018-08-21:
Hmm that's really bad :(

It would be very helpful if you could determine how to reproduce this bug.

> and it also didn't create automatic backups ... loads of work was lost that day :/

I'm sorry ;) You don't have the project still open? Probably you could just remove the two directories mentioned in the error message to successfully save the project...

SaculRennorb, 2018-08-21:
well i was planing on doing exactly that , but unfortunately i triggered #305 and well ... that's how i 'closed' the program and found out that the backup was corrupt / couldn't find a reference

ubruhin, 2018-08-21:
> well i was planing on doing exactly that , but unfortunately i triggered #305 and well 

Oh, what a bad luck :disappointed: 

You didn't modify/remove any project files (or executing Git commands like `git reset`) while the project was open, right?

I think it's not ideal that saving a project library element fails if the destination is an already existing empty directory, I will change it to allow saving into empty directories. But I don't know if this would also fix the issue you encountered (depends on whether the destination directory was empty or not). I really don't know how it can happen that the destination directory already exists :(

SaculRennorb, 2018-08-21:
maybe the bug occured because i worked onthe library and at the same time used elements from the lib inthe schematic / pcb designer? dont know if this type of 'parallel working' is intended, bit it seemed to sync every window just fine

ubruhin, 2018-08-22:
Good point. Actually this kind of parallel working should work without any issues, but maybe there is still a bug which can be triggered that way.

ubruhin, 2018-08-22:
OK now I can reproduce it:

1. Add a component to the schematic
2. Save the project
3. Remove the previously added component from the schematic
4. Add the same component again
5. Save the project -> Error as described above

SaculRennorb, 2018-08-22:
ough -  well you kinda have to do that if you make changes to a component in your lib, because the symbol isn't automatically updated in your schematic

PR #315 (2018-08-29): Refactor ProjectLibrary to fix several issues
- Add support for saving project libraries to backup directory "library~" instead of "library". Now the backup/restore feature of projects should also work properly for the project library. Fixes #47.
- Fix possible error when saving the project library after replacing a library element (remove element and add another one with same UUID). Fixes #312.
- Allow library elements to be saved into empty directories. Sometimes empty directories are not cleaned up properly (for example by Git), so we should handle empty directories just like non-existing d [... cut]
closed 2018-08-29 by ubruhin