LibrePCB/LibrePCB #313: Install icon to PREFIX/share/icons

Alexander-Wilms, 2018-08-25:
##### LibrePCB VERSION

81e7eb8

##### OS / ENVIRONMENT

Fedora 28
##### SUMMARY

`PREFIX/share/icons/hicolor/scalable/apps/librepcb.svg` and `PREFIX/share/icons/hicolor/scalable/mimetypes/` are the preferred location to install the icons, which also makes packaging easier.
[Freedesktop icon spec](https://standards.freedesktop.org/icon-theme-spec/icon-theme-spec-latest.html)
##### STEPS TO REPRODUCE

1. Build and install LibrePCB
##### EXPECTED RESULTS

Icon should be installed to `PREFIX/share/icons/hicolor/scalable/apps/librepcb.svg`
##### ACTUAL RESULTS

Icon is installed to `PREFIX/share/pixmaps/librepcb.svg`

ubruhin, 2018-08-26:
Thanks for pointing this out @Alexander-Wilms, sounds like a simple and reasonable change.

PR welcome :)

Alexander-Wilms, 2018-08-26:
I've created some similar patches for projects using cmake or autotools, so I'll try. However, the only place mentioning pixmaps is https://github.com/LibrePCB/LibrePCB/blob/68577ecf8f39299ef4d81ff964b01c3908d1f10b/dist/installer/packages/librepcb.nightly.app/meta/installscript.qs . Is that the right place to look?

ubruhin, 2018-08-26:
> I've created some similar patches for projects using cmake or autotools, so I'll try.

Great :smiley: 

> Is that the right place to look?

Basically the "share" directory is copied 1:1 to `PREFIX/`, so there is no *.pro file or something like that to adjust. Only the file `installscript.qs` depends on the images path, so yes, that's the only file to adjust.

commit fb2a8139 (2018-08-29): Use preferred icon paths and add appdata (#318)

* Add appdata (Fixes #314)
* Install icons to path preferred by Freedesktop spec (Fixes #313)
* Use reverse DNS naming for desktop file and icon
* Fix window icon under wayland
closed 2018-08-29 by ubruhin