LibrePCB/LibrePCB #335: SchematicEditor: Net labels are not mirrored properly

ubruhin, 2018-09-29:
##### LibrePCB VERSION
LibrePCB 0.1.0-unstable (ec7b25ecb2271ac19fb9c8bc4b48604d2b99cd58)

##### OS / ENVIRONMENT
All

##### SUMMARY
#268 added support for mirroring schematic elements, but netlabels are not mirrored properly. See https://github.com/LibrePCB/LibrePCB/pull/268#issuecomment-395874249 for details.

##### STEPS TO REPRODUCE
Select parts of a schematic (including netlabels) and mirror it.

[image]

##### EXPECTED RESULTS
Netlabels should be mirrored *exactly*.

##### ACTUAL RESULTS
Netlabels are mirrored, but have an offset then.

[image]

alfishe, 2019-05-12:
Issue has two aspects:
1) Labels have no clue about left-to-right render or right-to-left. Also there will be issues with Arabic languages. Potentially with some oriental (top-to bottom render) as well
2) Items abstraction does not allow access to on-screen dimensions (width in particular case)

alfishe, 2019-05-12:
Didn't get the idea why Net label (Text part) should be snapped to the grid. Probably it shouldn't like any other label positioned relatively to it's anchor element (that already snapped properly)

ubruhin, 2019-05-13:
> Didn't get the idea why Net label (Text part) should be snapped to the grid.

Basically because it allows consistent placement and thus better look of schematics. If netlabels are not snapped to grid, no exact placement is possible so they all have different distance to the netline (e.g. 0.1mm, 0.6mm, 0.3mm, ...) which looks ugly.

But of course we could implement a modifier (like SHIFT or so) to allow dragging netlabels (and maybe other elements) without snapping to grid.

alfishe, 2019-05-15:
Yeah, agree that there should be "snapped" mode by default and separate adjustment made for each element and it's labels individually (with all such offsets stored in meta-information). All "adult" CAE/CAD systems use such approach.

We're returning again to feature requirements / roadmap =)

commit 9be8db0a (2019-05-24): SchematicEditor: Fix net label offset after mirroring (#451)

Fixes #335
commit 41676a63 (2019-07-08): SchematicEditor: Fix net label offset after mirroring (#451)

Fixes #335
(cherry picked from commit 9be8db0a6dcff61ab6df9b872ad21b6728db3c7a)
closed 2019-05-24 by ubruhin