LibrePCB/LibrePCB #336: Cannot create new category in Library editor

ovidiub13, 2018-09-30:
##### LibrePCB VERSION
Version: 0.1.0-unstable
Git revision: ec7b25ecb2
Build date: 2018-09-29 13:56:00 (EEST)

##### OS / ENVIRONMENT
Linux, Fedora 28

##### SUMMARY
Cannot create new categories in the Library editor.

##### STEPS TO REPRODUCE
* Create new Library and edit it
* Add a component
* On the `Edit Metadata` screen, type in the `Category` the name of a new category

##### EXPECTED RESULTS
Simulation of new category hierarchy to be shown bellow the field. Just like it's shown when selecting an existing category.
Able to click `Next`, and new category to be created.

##### ACTUAL RESULTS
`Next` button disabled. Label below field just displays `Root category`.

P.S.: The choose category menu has no option to create a new category.

P.P.S.: The user NEVER cares to see the UUID of a component or a category.

ubruhin, 2018-09-30:
Hi @ovidiub13,

> On the Edit Metadata screen, type in the Category the name of a new category

You cannot enter the name of a category here, you have to use the "..." button to choose an existing category. The category needs to be created first, as described [in the user manual](https://docs.librepcb.org/#gettingstarted-libraries-cmpcat). Or you can leave it empty and set it afterwards.

Currently this is how it works, but I agree that this is a bit cumbersome ;)

> P.P.S.: The user NEVER cares to see the UUID of a component or a category.

I agree, probably I will hide all UUIDs in the library editor soon.

rnestler, 2018-10-01:
> I agree, probably I will hide all UUIDs in the library editor soon. EDIT: Done in #337

Isn't it still useful at this alpha/beta state of LibrePCB to still see which components have which UUID? I could imaging that it may help debugging if something goes wrong.

ubruhin, 2018-10-01:
> Isn't it still useful at this alpha/beta state of LibrePCB to still see which components have which UUID? I could imaging that it may help debugging if something goes wrong.

That's why I added a menu item in the library editor to open a library element in the file manager, there you also see the UUID etc.

And if needed, we can still some time add some kind of "developer mode" which shows more information of elements.

ubruhin, 2018-10-04:
With #337, the text field for the category UUID does no longer exist. Now only the label with the category name is shown:
[image]

I think now it should be less confusing, so I close this issue ;)

closed 2018-10-04 by ubruhin