LibrePCB/LibrePCB #339: Any critical changes needed before the first release?

ubruhin, 2018-09-30:
I guess the current state of the project is ready for the first release :smiley: 

Does someone think there are still things which need to be done before the first release?
- Very critical bugs?
- Missing extremely important features?
- Critical file format issues?
- Critical architecture issues?
- Something else?

But keep in mind: the first release (candidate) doesn't need to be perfect ;) Small bugs are not that bad, as long as they are not too annoying for the user. And of course features like a DRC would be nice, but that can be added in a later release...

And now: TEST, TEST, TEST, REVIEW, REVIEW, REVIEW! :smile:

dbrgn, 2018-10-15:
What about the PNS router? Is there a lot of work left to include that?

ubruhin, 2018-10-15:
That's not really critical ;)

And I'm still waiting until someone creates a standalone library for the PNS router (see https://github.com/LibrePCB/librepcb-rfcs/issues/1#issuecomment-413439122)...

ubruhin, 2018-10-20:
No critical issues reported -> start creating the release :)

closed 2018-10-20 by ubruhin