LibrePCB/LibrePCB #343: Schematic Editor: Placement mode exited by right click + move

dbrgn, 2018-10-06:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.0-unstable
Git Revision:     4dd24a39b702868b6f364eb4ae3734c6983495f8
Build Date:       2018-10-05T20:34:10
Qt Version:       5.9.5 (built against 5.9.5)
CPU Architecture: x86_64
Operating System: Windows 10 (10.0)
Platform Plugin:  windows
```

##### SUMMARY

When placing symbols on the schematic the symbol can be rotated using right click. When moving the mouse pointer while the right mouse button is clicked, the placement mode is exited.

##### STEPS TO REPRODUCE

1. Place symbol
2. Click the right mouse button quickly, this should result in the symbol being rotated
3. Now click and hold the right mouse button and move the mouse pointer. The symbol disappears.

##### EXPECTED RESULTS

The placement mode should only be exited with ESC

dbrgn, 2018-10-06:
I'm marking this as easy since I suspect that it's only a Qt signal handler issue. @ubruhin can you confirm that?

ubruhin, 2018-10-06:
Probably it's a bug in our schematic editor state machine, then it must be in [`SES_AddComponent`](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp).

Actually for this kind of bugs it would be very nice to add functional tests to make sure the same bug never occurs again. But writing such a test is probably much harder than fixing the bug ;)

PR #761 (2020-08-31): Schematic editor: Fix leaving tool with right click + move
When pressing the right mouse button while moving the cursor, each schematic editor tool unintentionally exited because the event was not handled by the tool. So the event was handled by the state machine, which switches to the last active tool by the right mouse button click.

Now the event is accepted by the corresponding tools to avoid this issue.

Fixes #343.
closed 2020-08-31 by ubruhin