LibrePCB/LibrePCB #344: Schematic Editor: Allow placing symbol with double click

dbrgn, 2018-10-06:
##### SUMMARY

When listing symbols, allow placing a certain symbol by double clicking it. Right now, the user has to click "OK".

[image]

robbawebba, 2018-10-06:
@dbrgn I would like to give this a shot! This would be the first Qt-based project I'll work on, so I would like to run my thoughts by you to be sure I've got the correct approach.

Currently the "OK" and "Cancel" buttons fulfill the `accept()` and `reject()` slots of the containing `AddComponentDialog`. When double-clicking an item in the tree view, the item generates an `itemDoubleClicked` signal. So we'll want to handle that signal and have it mimic the behavior of the "OK" button? So this signal will also fill the `accept()` slot of `AddComponentDialog`?

ubruhin, 2018-10-06:
> @dbrgn I would like to give this a shot! This would be the first Qt-based project I'll work on, so I would like to run my thoughts by you to be sure I've got the correct approach.

Great! :smiley: 

> Currently the "OK" and "Cancel" buttons fulfill the accept() and reject() slots of the containing AddComponentDialog. When double-clicking an item in the tree view, the item generates an [...]

Yes, sounds good so far :)

commit 7c0535b2 (2018-10-08): AddComponentDialog: place a component by double-clicking list item

To place a symbol from the 'Add Component' dialog, the user currently
has to select the desired symbol from the list and click 'OK'. This
change allows users to place a symbol by double-clicking the symbol in
the list without clicking the 'OK' button.

Fixes #344
PR #346 (2018-10-08): AddComponentDialog: place a component by double-clicking list item
To place a symbol from the 'Add Component' dialog, the user currently
has to select the desired symbol from the list and click 'OK'. This
change allows users to place a symbol by double-clicking the symbol in
the list without clicking the 'OK' button.

Fixes #344
closed 2018-10-08 by ubruhin