LibrePCB/LibrePCB #348: Library Editor: Allow to display and copy UUIDs

ubruhin, 2018-10-10:
##### SUMMARY
UUIDs were removed from the UI in #337 to avoid confusion of users. But they are sometimes still needed, for example when creating elements by scripts (e.g. [librepcb-parts-generator](https://github.com/LibrePCB/librepcb-parts-generator)). So there should be a way to display UUIDs (and copy them to clipboard) of library elements, symbol pins, footprint pads etc.

##### IMPLEMENTATION
I'm not sure how this could be implemented without cluttering the UI. It's important that UUIDs are normally not visible.

dbrgn, 2018-10-11:
What about a menu entry next to the "open in file manager" entry called either "developer information" or "details"? It could contain the own UUID, the category UUID, etc. And maybe even the source code.

ubruhin, 2018-10-11:
Yeah that would be an option. But there are much more UUIDs than only the library element's UUID: Category, Component Signals, Component Symbol Variants, Component Symbol Variant Items, Package Pads, Polygons, Drills, Texts etc.

dbrgn, 2018-10-11:
Yeah of course, you could display them in a tree like structure.

dbrgn, 2022-07-17:
How about an entry in the context menu called "Copy UUID to Clipboard"?

[image]