LibrePCB/LibrePCB #352: Allow to choose or create a workspace when a one is already opened

ubruhin, 2018-10-15:
##### SUMMARY
Currently the "first run wizard" is the only way to create a new workspace interactively (asking for location and initial settings). After opening an existing workspace, it's no longer possible to create a new workspace interactively. The only way is to choose "File" -> "Switch Workspace" in the control panel and choose an empty directory, but that's not intuitive and doesn't allow to set the initial workspace settings.

##### IMPLEMENTATION

I think it would be nice if the "Switch Workspace" menu item just shows the "first run wizard", but without its first page ("Welcome to LibrePCB"). That way the user could interactively choose an existing or create a new workspace. Maybe the `FirstRunWizardDialog` should be renamed to `WorkspaceChooserDialog` or so, and the visibility of the welcome page should be configurable.

rummanwaqar, 2018-10-17:
I can take this issue

ubruhin, 2018-10-17:
@rummanwaqar cool, looking forward for a pull request! :smiley: Don't hesitate to ask if you have any questions :)

PR #688 (2020-04-22): ControlPanel: Use workspace chooser wizard to switch workspace
Instead of a simple path input box, now the full-featured first-run-wizard is shown when requesting to switch the workspace by the ControlPanel menu item File -> Switch Workspace. See #352 for details

This was now easy to implement because in #652 the settings initialization page was moved into a separate wizard.

Fixes #352
closed 2020-04-22 by ubruhin