LibrePCB/LibrePCB #355: Windows too small by default on HiDPI displays on Windows

prokoudine, 2018-10-22:
Windows 10, 2560x1440px display.

Default window after start-up:

[image]

Default project window after creating a project:

[image]

That's a tad too small :)

ubruhin, 2018-10-23:
Hi @prokoudine! But this happens only when starting LibrePCB the very first time, right?

I'm not sure if and how this could be fixed easily. Probably it would require some logic to detect if the window size is available in the settings, and if not it should be initialized with a default value. But default values which are fine on a HiDPI display are probably too large for a small display ;)

prokoudine, 2018-10-23:
Yes, only the first time you start LibrePCB.

Something I know GIMP does for its splash screen is setting its width as 1/2 of the screen width. I'm guessing you could do something like this. E.g. detect display width and set the window width to 3/4 of that. That would work for both LoDPI and HiDPI displays.

Not sure if Qt has an equivalent of _gdk_screen_get_width_ and _gdk_screen_get_height_, but maybe it's worth checking for?

ubruhin, 2018-10-24:
> Something I know GIMP does for its splash screen is setting its width as 1/2 of the screen width. I'm guessing you could do something like this. E.g. detect display width and set the window width to 3/4 of that. That would work for both LoDPI and HiDPI displays.

Yeah that would be an option.

> Not sure if Qt has an equivalent of gdk_screen_get_width and gdk_screen_get_height, but maybe it's worth checking for?

I think [`QGuiApplication::primaryScreen()`](https://doc.qt.io/qt-5/qguiapplication.html#primaryScreen-prop) can be used to obtain the screen size.

So probably it wouldn't be too hard to implement that :)

Following windows exist currently and thus would need to be adjusted:
- ControlPanel
- LibraryEditor
- SchematicEditor
- BoardEditor

PR #1525 (2025-05-18): Initially maximize new main windows
When opening LibrePCB (or additional main windows) the very first time, maximize the window because I think this is the most reasonable default window size.

Closes #355
closed 2025-05-18 by ubruhin