LibrePCB/LibrePCB #357: UUID Checks

ouabache, 2018-10-26:
I am trying to write a script that will convert a Kicad library into a LibrePCB one but I am stuck with generating my own UUIDs. They look ok buy I get:

[EXCEPTION] String is not a valid UUID: "f21861ce-a59a-1934-a78f-c02195363aaf" (../../../../libs/librepcb/common/uuid.cpp:68)
[EXCEPTION] File parse error: String is not a valid UUID: "f21861ce-a59a-1934-a78f-c02195363aaf"

Is this a valid UUID?

dbrgn, 2018-10-26:
In UUIDs, [the 13th character indicates the UUID vesion](https://en.wikipedia.org/wiki/Universally_unique_identifier#Format):

    f21861ce-a59a-1934-a78f-c02195363aaf
                  ^

In your case, this seems to be a [UUID 1](https://en.wikipedia.org/wiki/Universally_unique_identifier#Version_1_.28date-time_and_MAC_address.29) (date-time and MAC based).

Note: "Usage of the node's network card MAC address for the node id means that a version 1 UUID can be tracked back to the computer that created it." Also: "In contrast to other UUID versions, version 1 and 2 UUIDs based on MAC addresses from network cards rely for their uniqueness in part on an identifier issued by a central registration authority".

As far as I know, LibrePCB uses [Version 4 UUIDs](https://en.wikipedia.org/wiki/Universally_unique_identifier#Version_4_.28random.29) (random). For example [this package](https://github.com/LibrePCB-Libraries/LibrePCB_Base.lplib/blob/master/pkg/11fe8774-ade2-4b36-bde6-1ec36ae627df/package.lp) has this UUID:

    11fe8774-ade2-4b36-bde6-1ec36ae627df
                  ^

In Python, generating a UUID4 is as simple as `import uuid; uuid.uuid4()`. But it depends on your generator of course.

@ubruhin as long as a UUID is a valid UUID of any type, shouldn't we simply parse it?

ouabache, 2018-10-26:
I thought that librePCB used random but if that's not the case then we need
to document it.

dbrgn, 2018-10-26:
LibrePCB *does* use random (type 4) UUIDs, but that does not mean that all hex characters are random :slightly_smiling_face: Valid UUIDs need to adhere to the specification.

The main question here is whether we *only* support type 4 UUIDs, or not. That's up to @ubruhin to decide.

ubruhin, 2018-10-26:
Yes, we only support type 4 (random) UUIDs and I see no reason to support other types ;)

ouabache, 2018-10-26:
So where is the  specification? I tried forcing char 13 to 4 but it still doesn't accept them.

dbrgn, 2018-10-26:
I think here's the specification: https://tools.ietf.org/html/rfc4122 The Wikipedia article linked above also helps.

But how are you generating these UUIDs? If you use a programming language to generate the library elements, you can use a UUID library. If you write them by hand, you can use something like https://www.uuidgenerator.net/version4.

ouabache, 2018-10-26:
I am calling rand(65536) 8 times in perl and stuffing them all together. Looks like I need to do some more research.

dbrgn, 2018-10-26:
I'm not familiar with the Perl ecosystem at all, but maybe there's something for you on CPAN: https://metacpan.org/search?q=uuid

ouabache, 2018-10-26:
Thanks, I had found it on CPAN but since dropbox is going to drop support for my system in 12 days I need to reimage my machine to 18.10 and put my dropbox on a bigger drive. I will add UUID support at that time.

I am beginning to like LibrePCB. The more I use it the  easier it becomes.

John Eaton

ubruhin, 2018-11-25:
I guess this is solved(?). If not, feel free to reopen :)

closed 2018-11-25 by ubruhin

vicencb, 2023-12-30:
Hi,
i would like to reopen this issue because i think it would be very useful to support type 3 or 5 UUIDs.
I've been playing around with a library generator and implemented it with type 5 UUIDs generated deterministically and hierarchically like this:
```
librepcb_namespace = new_uuid_sha1("00000000-0000-0000-0000-000000000000", "LibrePCB")
  cust_lib_gen_ns = new_uuid_sha1(librepcb_namespace, "custom_library_generator")
    cmpcat_uuid = new_uuid_sha1(cust_lib_gen_ns, "cmpcat")
    pkgcat_uuid = new_uuid_sha1(cust_lib_gen_ns, "pkgcat")
    cmp_uuid    = new_uuid_sha1(cust_lib_gen_ns, "cmp"   )
    pkg_uuid    = new_uuid_sha1(cust_lib_gen_ns, "pkg"   )
      some_pkg_uuid = new_uuid_sha1(pkg_uuid, "package name")
        ... more hierarchy levels ...
    dev_uuid    = new_uuid_sha1(cust_lib_gen_ns, "dev"   )
    sym_uuid    = new_uuid_sha1(cust_lib_gen_ns, "sym"   )
```
and this is simpler than storing loads of name/uuid pairs in order to achieve reproducibility.

ubruhin, 2024-01-02:
Hi,

> I've been playing around with a library generator and implemented it with type 5 UUIDs generated deterministically and hierarchically like this:

Generally this is interesting and I agree it would be simpler than storing name/uuid pairs. However, I'm not 100% convinced it is really a good idea. One drawback is the increased risk of UUID conflicts since people might use the same concept of generating UUIDs (especially when copy&pasting an existing generator). Another drawback is that you must never change the keys used for UUID generation. In `librepcb-parts-generator` from time to time we realize that we should rename something (for example if some name is not unique anymore), which can be done by manually updating the name/uuid pairs accordingly. If UUIDs are generated directly from some input strings, these input strings must never be changed.

Thus at the moment I still think it's much safer to accept only random UUIDs.