LibrePCB/LibrePCB #360: Linux keyboard layout wrong

ruabmbua, 2018-10-27:
##### VERSION / OS / ENVIRONMENT

```
4.18.16-arch1-1-ARCH #1 SMP PREEMPT Sat Oct 20 22:06:45 UTC 2018 x86_64 GNU/Linux
LibrePCB Version: 0.1.0-rc1
Git Revision:     d49c0a70e23f5690e4d5785684ef828e04c2cb26
Build Date:       2018-10-21T12:05:18
Qt Version:       5.9.1 (built against 5.9.1)
CPU Architecture: x86_64
Operating System: Antergos Linux
Platform Plugin:  xcb
...
```

##### SUMMARY

Keyboard input does not reflect operating system settings. I am using a german QWERTZ keyboard, but the input is registered as QWERTY. No other application on my system has this issue. 

##### STEPS TO REPRODUCE
Type `z` in any of the text fields using a german keyboard layout.

##### EXPECTED RESULTS
See a `z`.

##### ACTUAL RESULTS
Got a `y`.

##### ADDITIONAL INFORMATION

ruabmbua, 2018-10-27:
Just launched it from the shell, nothing suspicious:

```
QApplication: invalid style override passed, ignoring it.
Load stroke font: "newstroke.bene"
Start loading font "/home/rruckerbauer/Anwendungen/LibrePCB/stable/share/librepcb/fontobene/newstroke.bene"
[  INFO   ] "LibrePCB 0.1.0-rc1 (d49c0a70e23f5690e4d5785684ef828e04c2cb26)" (:0)
[  INFO   ] "Qt version: 5.9.1 (compiled against 5.9.1)" (:0)
[  INFO   ] "Resources directory: /home/rruckerbauer/Anwendungen/LibrePCB/stable/share/librepcb" (:0)
[  INFO   ] "Application settings: /home/rruckerbauer/.config/LibrePCB/LibrePCB.ini" (:0)
[DEBUG-MSG] Started network access manager thread. (:0)
[DEBUG-MSG] Load workspace settings... (:0)
[DEBUG-MSG] Workspace settings successfully loaded! (:0)
[DEBUG-MSG] Load remote workspace library: "1fcb0725-d2e8-4f5d-892c-d1efb44d2963.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "326f091b-b715-44bf-b385-b613cd60d9f3.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "6ccc516c-21b7-4cd5-9cf2-7a04cfa361c6.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "a9ddf0c6-9b1c-4730-b300-01b4f192ad40.lplib" (:0)
[DEBUG-MSG] Load remote workspace library: "e3ce49bf-e959-4092-8b48-aeead756f8d1.lplib" (:0)
[DEBUG-MSG] Load workspace library database... (:0)
[DEBUG-MSG] Workspace library database successfully loaded! (:0)
[DEBUG-MSG] Successfully loaded font "/home/rruckerbauer/Anwendungen/LibrePCB/stable/share/librepcb/fontobene/newstroke.bene" with 2573 glyphs (:0)

```

ruabmbua, 2018-10-27:
Just built master from source, and it works. Maybe its a problem with the 0.1rc1 build, or there has already been a fix.

ubruhin, 2018-10-27:
> Just launched it from the shell, nothing suspicious:

Yep, looks good :+1:

> Just built master from source, and it works.

Interesting, so it must be either an issue with the Qt version (5.9.1) deployed together with our binaries or with the deployment itself. Which Qt version do you use when building from source?

> Maybe its a problem with the 0.1rc1 build, or there has already been a fix.

On master there is no fix for that. You could build the "0.1.0-rc1" tag (same version as the release binaries), I think it will also work.

ruabmbua, 2018-10-27:
##### About dialog from master build:

```
LibrePCB Version: 0.1.0-unstable
Git Revision:     65802d7e2045425307df74feaaba6507dfe6b276
Build Date:       2018-10-27T18:57:15
Qt Version:       5.11.2 (built against 5.11.2)
CPU Architecture: x86_64
Operating System: Antergos Linux
Platform Plugin:  xcb
```

dbrgn, 2018-10-27:
@ruabmbua you used the AppImage, right?

I'm on Arch. I just downloaded it (about dialog details are identical, including build date) and cannot reproduce your issue. My keyboard is set to Swiss German, which seems to work correctly in LibrePCB.

ruabmbua, 2018-10-27:
No, I downloaded the installer not the appimage.

ruabmbua, 2018-10-28:
Same problem with appimage, I just verified it.

ruabmbua, 2018-10-28:
I just built the release branch from source ant then did `sudo make install`.
The installation does now recognize my input method.
##### BUILD INFO:
```
LibrePCB Version: 0.1.0-rc1
Git Revision:     d49c0a70e23f5690e4d5785684ef828e04c2cb26
Build Date:       2018-10-28T13:03:16
Qt Version:       5.11.2 (built against 5.11.2)
CPU Architecture: x86_64
Operating System: Antergos Linux
Platform Plugin:  xcb
```

ubruhin, 2018-10-28:
Could you quickly try this AppImage, which is built with Qt 5.10.1? https://download.librepcb.org/nightly_builds/travis-qt-5_10_1/librepcb-nightly-linux-x86_64.AppImage

ruabmbua, 2018-10-28:
Also not working.
Is it possible, that the problem is related to XWayland? Maybe I should try the official build form a native X session.

ruabmbua, 2018-10-28:
Just tried it with X11, and now it works. Seems like my local QT version, with which I build the application somehow fixes the input problem with XWayland / libinput...

ubruhin, 2018-10-28:
Hmm okay... So I really don't know what we can do to fix this issue :(

ruabmbua, 2022-04-14:
I think this can be closed, since Antergos does not exist any more ;-)

closed 2022-04-14 by ruabmbua