LibrePCB/LibrePCB #362: Cannot change from millimeters to inches

ouabache, 2018-10-27:
##### VERSION / OS / ENVIRONMENT

LibrePCB Version: 0.1.0-rc1
Git Revision:     d49c0a70e23f5690e4d5785684ef828e04c2cb26
Build Date:       2018-10-21T12:05:18
Qt Version:       5.9.1 (built against 5.9.1)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.1 LTS
Platform Plugin:  xcb

```

```

##### SUMMARY
changing librePCB to inches has no effect on library or schematic editors

##### STEPS TO REPRODUCE
change workspace to inches

##### EXPECTED RESULTS
editors should configure to inches

##### ACTUAL RESULTS
editors stay in millimeters

##### ADDITIONAL INFORMATION

ubruhin, 2018-10-28:
This is simply not implemented yet ;)

I think we should use this workspace setting only when creating new projects. Each schematic page and board stores the grid interval and unit in their files separately. This allows for example to use grid 0.1 Inches for schematics, but 1.0mm for boards.

But when creating a new project, its schematic and board should be initialized with the unit defined in the workspace settings.

In addition, the library editor should always use the unit defined in workspace settings (in contrast to schematics/board, grid interval is not stored in library elements).

ouabache, 2018-10-28:
So will you have libraries in either mm or inch and convert when used in the other domain?

Then you should have every symbol file declare its snap grid spacing in whatever units that it uses. Add a new field (snapgrid  2.54) to every symbol and you you will know that it is in metric. Once you add inches then you know to scale all the numbers before use. If you see (snapgrid 0.1) then you know it is in inches and you should scale if you are using metric.

This will also also avoid the problem that kicad often sees when you build a library using a snapgrid smaller than the schematic editor uses. Your pins can't connect.  This way the pins will always scale to whatever snapgrid the editor is using. The graphics may change size but the connectivity is intact.

You really do want to add this before your first official release. It may never get used but that's better than trying to deal with legacy code if you have to add it later.

John Eaton

ubruhin, 2018-10-28:
I don't think symbols and footprints need to know what grid interval was used to draw them. The snap issue wouldn't be solved with this, it's an issue related only to the editors. Board and schematic editors must work properly, independent of the grid interval used in symbols and footprints.

In files, LibrePCB always uses millimeters for everything. Switching to inches is only a matter of editors, they just convert all values between millimeters and inches.

ouabache, 2018-10-28:
Ok, That answers my questions. Symbols are always stored metric and the editor must handle the conversion and all connection hot spots must fall on a 2.54 mm grid  or they won't work.

John Eaton

ubruhin, 2018-10-28:
> Symbols are always stored metric and the editor must handle the conversion

Correct.

> and all connection hot spots must fall on a 2.54 mm grid or they won't work.

Currently that's true for the schematic editor, but this limitation can be fixed with a more sophisticated snap logic.

But Footprints do (of course) work even if pads are not on the 2.54mm grid. The board editor also snaps to pads which are not on the grid, although it's not yet working very well (currently there must be at least one point of the pad's **area** on the board's grid). But of course this could also be improved ;)

SweRavn, 2019-01-03:
@ubruhin: Hi, I just found out about this Great project and would like to contribute. I have downloaded and compiled everything and can run from Qt Creator. I was thinking this issue could be a good place to start. Seems like a reasonably small thing. But I guess there might be several places that will need conversion. Perhaps adding a method to the settings that gives back the coordinate in current units. Is that a sensible way? Would be nice maybe to ask a Point (Length) that directly, but might add a dependency in the wrong direction or?
Further, I found the place in the Library editor where units are selected, but I could not find it in the Schematics or Board editors. Where should this setting be done for those editors? Should I add a settings dialog for those editors?

@ouabache: I was considering looking at this issue if you did not already start with it. Have you?

ubruhin, 2019-09-20:
Btw, #519 added the widget `LengthEdit` which is able to display/edit `Length` values in the desired unit. So we should replace all `QDoubleSpinBox` widgets which display a length by `LengthEdit` and call the `setUnit()` method to select the proper unit. Then we don't have to care about the calculations for unit conversion.

plocher, 2020-01-25:
Jumping in late...
As a 'merican, I may well be in the minority here, but ... :-)

It would be "really nice" if the units concept was finer grained than just "schematic" or "board" editor, because some things override these global preferences.  

For example, drill sizes are uniformly metric and just don't make sense when expressed in mils, schematics are required to be in mils with 0.100 grid, while routing traces between package leads requires the traces to align to the package's pitch - or a DRC-related division thereof (1/2 with 1 trace between pads, 1/3 for two traces... limited by the DRC minimum trace sizes and spacing).  These things should be easy and automatic by default, even if/when the more generic settings are chosen by the user.

ubruhin, 2020-04-12:
I started working on this in #684. Unfortunately, working with units other than Millimeters is a pain :sob:

## Problem 1: Inches vs. Mils

One problem is that for different kinds of length values, different units are desired (e.g. inches for coordinates, but mils for clearance values). I think it would be a nightmare to make this fully configurable. Since with #684 all length input fields can be changed to a different unit, I think the most reasonable solution would be to simply memorize the last set unit of each field, and restore them when opening the next time. Still a bit painful to implement, but probably it solves the issue pretty well. In the end, the unit configured in workspace settings is then only used as fallback for fields which don't provide a memorized unit yet (e.g. when opening LibrePCB the first time).

I wonder if it even makes more sense to only choose between "metric" and "imperial" system in the workspace settings (instead of exact units like "inches" or "mils"). Then LibrePCB could choose a reasonable unit (of the configured measurement system) for each input field separately. For example with the imperial system, coordinate input fields are set to inches, while DRC clearances are set to mils.

## Problem 2: Unit depends on use-case

As @plocher wrote:

> It would be "really nice" if the units concept was finer grained than just "schematic" or "board" editor, because some things override these global preferences.
>
> For example, drill sizes are [...]

Another example is that our library conventions specify various values in multiple of 0.1mm (text height, line width, ...). If the corresponding input fields represent the value in the unit configured in the workspace, odd values are displayed (e.g. 3.937mils instead of 0.1mm) and it's impossible to compare them with the conventions without switching to millimeters. I see two possible solutions:
- Ignore this issue. When memorizing the unit of input fields is implemented, an (American) user has to manually switch the corresponding input fields to millimeters only once, so it's not *that* annoying.
- Ignore the configured workspace unit for specific input fields. For example the "line width" input field could be initialized with millimeters, even if the workspace unit it set to inches. Might be a good solution for most users, but somehow I don't like it because it feels like "magic". Americans might even think it's a bug if an input field is set so millimeters, even though they configured inches as default(?).

Ideas and opinions welcome :slightly_smiling_face:

dbrgn, 2020-04-13:
> I think the most reasonable solution would be to simply memorize the last set unit of each field, and restore them when opening the next time

I think that does sound reasonable, as long as the default can be set somehow (metric or imperial).

> I wonder if it even makes more sense to only choose between "metric" and "imperial" system in the workspace settings (instead of exact units like "inches" or "mils").

Hm, maybe. On the other hand, if you're working with small PCBs with SMD parts you might prefer mils and if you work with large PCBs with THT parts you might prefer inches. If you can only choose "imperial", then you'll have to change the default for every field you're editing.

But that's probably something that should be answered by the people that use imperial units regularly :slightly_smiling_face: 

(Also, strictly speaking the US does not use the imperial system, it's called [US Customary Units](https://en.wikipedia.org/wiki/United_States_customary_units).)

> Ignore this issue. When memorizing the unit of input fields is implemented, an (American) user has to manually switch the corresponding input fields to millimeters only once, so it's not that annoying.

I think that would be fine. After all, 192 out of 195 countries in the world use the metric system, and Myanmar and Liberia are on their way to fully move to the metric system :stuck_out_tongue:

plocher, 2020-04-13:
So, what measurement settings do "non-US-Customary Unit" users use for
schematics?  I find it hard to imagine that anyone is comfortable with
managing the multiples of 0.100 inches in "that other system" :-)
IMO, "1.27", "2.54" and "5.08" are as awkward as customary imperial drill
sizes are for US ...

It would be great to have a global/startup setting like "Usual and
Customary" that set inches for the schematic and symbol grids and used
metric for everything else.

  -John

ubruhin, 2020-04-13:
> So, what measurement settings do "non-US-Customary Unit" users use for
schematics?  I find it hard to imagine that anyone is comfortable with
managing the multiples of 0.100 inches in "that other system" :-)
IMO, "1.27", "2.54" and "5.08" are as awkward as customary imperial drill
sizes are for US ...

Converting multiples of 0.1 inches to millimeters is easy (multiplying with 25.4mm adds only one additional decimal place). So yes, users of the metric system are very familiar with 2.54mm and multiple of it, and therefore also use millimeters in schematics. We do not use inches or mils at all in an EDA tool.

The "real" problem is the other way around: Converting metric values (e.g. a multiple of 0.1mm) to inches requires to divide by 25.4, which almost always ends up with an infinite amount of decimal places - these numbers are a pain to work with, so Americans might consider switching to the metric system in this case :slightly_smiling_face:

dbrgn, 2020-04-13:
> I find it hard to imagine that anyone is comfortable with managing the multiples of 0.100 inches in "that other system" :-) IMO, "1.27", "2.54" and "5.08" are as awkward as customary imperial drill sizes are for US ...

I actually do use `1.27` and its multipliers a lot. After a while you have memorized those awkward values (and their typical multiples). But you are right that they're quite awkward to use. Being able to change the input field unit type would be very nice.

(Please don't take my remark about countries using non-metric systems seriously. I'm well aware that a lot of electronic components are based on inch-based spacing and that an EDA tool needs to support those :slightly_smiling_face:)

ubruhin, 2020-04-20:
OK thanks for your inputs :+1: I updated #684 as following:
- By default, all input fields use the unit defined in workspace settings, no matter if the numbers represent a drill size, coordinate, or whatever.
- When switching individual input fields manually to a different unit, it is saved in the local user settings and restored when opening the window the next time.

ubruhin, 2020-04-22:
Fixed in #684.

closed 2020-04-22 by ubruhin