LibrePCB/LibrePCB #364: Random CI failures on MacOS

ubruhin, 2018-10-29:
Functional tests randomly fail on MacOS CI, see for example https://travis-ci.org/LibrePCB/LibrePCB/jobs/447843334:

```
============================= test session starts ==============================
platform darwin -- Python 2.7.15, pytest-3.6.4, py-1.7.0, pluggy-0.7.1 -- /usr/local/opt/python@2/bin/python2.7
cachedir: tests/funq/.pytest_cache
rootdir: /Users/travis/build/LibrePCB/LibrePCB/tests/funq, inifile: setup.cfg
collected 30 items                                                             
tests/funq/commandline/test_open_project.py::test_open_project PASSED    [  3%]
tests/funq/controlpanel/test_create_project.py::test_new_project_wizard PASSED [  6%]
tests/funq/controlpanel/test_open_librarymanager.py::test_using_button PASSED [ 10%]
tests/funq/controlpanel/test_open_librarymanager.py::test_using_menu PASSED [ 13%]
tests/funq/controlpanel/test_open_project.py::test_open_dialog PASSED    [ 16%]
tests/funq/libraryeditor/test_copy_component.py::test PASSED             [ 20%]
tests/funq/libraryeditor/test_copy_component_category.py::test PASSED    [ 23%]
tests/funq/libraryeditor/test_copy_device.py::test PASSED                [ 26%]
tests/funq/libraryeditor/test_copy_package.py::test PASSED               [ 30%]
tests/funq/libraryeditor/test_copy_package_category.py::test PASSED      [ 33%]
tests/funq/libraryeditor/test_copy_symbol.py::test PASSED                [ 36%]
tests/funq/libraryeditor/test_create_component.py::test PASSED           [ 40%]
tests/funq/libraryeditor/test_create_component_category.py::test PASSED  [ 43%]
tests/funq/libraryeditor/test_create_device.py::test PASSED              [ 46%]
tests/funq/libraryeditor/test_create_package.py::test PASSED             [ 50%]
tests/funq/libraryeditor/test_create_package_category.py::test PASSED    [ 53%]
tests/funq/libraryeditor/test_create_symbol.py::test PASSED              [ 56%]
tests/funq/libraryeditor/test_open.py::test_metadata PASSED              [ 60%]
tests/funq/librarymanager/test_create_local_library.py::test PASSED      [ 63%]
tests/funq/librarymanager/test_install_remote_libraries.py::test PASSED  [ 66%]
tests/funq/projecteditor/test_add_component_dialog.py::test_if_dialog_is_active_widget PASSED [ 70%]
tests/funq/projecteditor/test_add_component_dialog.py::test_init_state PASSED [ 73%]
tests/funq/projecteditor/test_add_component_dialog.py::test_if_cancel_closes_dialog PASSED [ 76%]
tests/funq/projecteditor/test_add_component_dialog.py::test_if_accept_does_nothing_if_no_component_selected PASSED [ 80%]
tests/funq/projecteditor/test_add_component_dialog.py::test_select_component_with_one_symbvar PASSED [ 83%]
tests/funq/projecteditor/test_add_component_dialog.py::test_select_component_with_two_symbvars PASSED [ 86%]
tests/funq/projecteditor/test_add_component_dialog.py::test_select_device FAILED [ 90%]
tests/funq/projecteditor/test_add_component_dialog.py::test_select_component_by_doubleclick PASSED [ 93%]
tests/funq/projecteditor/test_add_component_dialog.py::test_select_device_by_doubleclick PASSED [ 96%]
tests/funq/projecteditor/test_open.py::test_window_titles PASSED         [100%]
=================================== FAILURES ===================================
______________________________ test_select_device ______________________________
add_component_dialog = <projecteditor.test_add_component_dialog.AddComponentDialogHelper object at 0x10fb951d0>
helpers = <conftest.Helpers object at 0x10fb4d450>
    def test_select_device(add_component_dialog, helpers):
        """
        Test selecting a device
        """
        # Select category
        category_tree = add_component_dialog.app.widget('schematicEditorAddComponentDialogCategoryTree')
        helpers.wait_for_model_items_count(category_tree, 1)
        category = category_tree.model_items().items[0]
        assert category.value == 'Capacitors'
        category.select()
    
        # Select device
        component_tree = add_component_dialog.app.widget('schematicEditorAddComponentDialogComponentTree')
        helpers.wait_for_model_items_count(component_tree, 1)
        device = component_tree.model_items().items[0].items[1]
        assert device.value == 'C-0805'
        device.select()
    
        # Symbol variant must be visible
        symbvar_combobox = add_component_dialog.app.widget('schematicEditorAddComponentDialogSymVarCombobox')
        helpers.wait_for_model_items_count(symbvar_combobox, 2, 2)
>       symbvar_items = symbvar_combobox.model_items()
tests/funq/projecteditor/test_add_component_dialog.py:161: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../Library/Python/2.7/lib/python/site-packages/funq/models.py:759: in model_items
    combo_edit_view = self.client.widget(path=self.path + internal_qt_name)
../../../Library/Python/2.7/lib/python/site-packages/funq/client.py:261: in widget
    widget.wait_for_properties(props)
[... 53 lines cut ...]
[DEBUG-MSG] Load stroke font: "newstroke.bene" (:0)
[DEBUG-MSG] Start loading font "/private/var/folders/nz/vv4_9tw56nv9k3tkvyszvwg80000gn/T/pytest-of-travis/pytest-1/test_select_device0/data/fixtures/Empty Project/resources/fontobene/newstroke.bene" (:0)
[DEBUG-MSG] load project metadata... (:0)
[DEBUG-MSG] metadata successfully loaded! (:0)
[DEBUG-MSG] load settings... (:0)
[DEBUG-MSG] settings successfully loaded! (:0)
[DEBUG-MSG] load project library... (:0)
[DEBUG-MSG] successfully loaded 0 symbols (:0)
[DEBUG-MSG] successfully loaded 0 packages (:0)
[DEBUG-MSG] successfully loaded 0 components (:0)
[DEBUG-MSG] successfully loaded 0 devices (:0)
[DEBUG-MSG] project library successfully loaded! (:0)
[DEBUG-MSG] load circuit... (:0)
[DEBUG-MSG] circuit successfully loaded! (:0)
[DEBUG-MSG] 1 schematics successfully loaded! (:0)
[DEBUG-MSG] 1 boards successfully loaded! (:0)
[DEBUG-MSG] project successfully loaded! (:0)
[DEBUG-MSG] Successfully loaded font "/private/var/folders/nz/vv4_9tw56nv9k3tkvyszvwg80000gn/T/pytest-of-travis/pytest-1/test_select_device0/data/fixtures/Empty Project/resources/fontobene/newstroke.bene" with 2573 glyphs (:0)
--------------------------- Captured stderr teardown ---------------------------
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Stopped network access manager thread. (:0)
[DEBUG-MSG] closed project: "/private/var/folders/nz/vv4_9tw56nv9k3tkvyszvwg80000gn/T/pytest-of-travis/pytest-1/test_select_device0/data/fixtures/Empty Project/Empty Project.lpp" (:0)
[DEBUG-MSG] Exit application with code 0 (:0)
==================== 1 failed, 29 passed in 156.26 seconds =====================
```

ubruhin, 2018-11-07:
This is *very* annoying, CI failures all the time :sob:  Someone interested in fixing this?

HBadertscher, 2018-11-16:
I am trying to reproduce this locally on my Macbook with OS X 10.13.6. I compiled the current master a2152fe52 and am running the tests in a virtual environment using  Python 2.7.10. I am using Qt 5.6.1 installed with Homebrew.

All tests are `PASSED`, but for the following tests:
- `tests/funq/projecteditor/test_add_component_dialog.py::test_select_component_with_one_symbvar`
- `tests/funq/projecteditor/test_add_component_dialog.py::test_select_component_with_one_symbvar`
- `tests/funq/projecteditor/test_add_component_dialog.py::test_select_device`
- `tests/funq/projecteditor/test_add_component_dialog.py::test_select_component_by_doubleclick`
- `tests/funq/projecteditor/test_add_component_dialog.py::test_select_device_by_doubleclick`

I get the following message *after* the test is completed and `PASSED`:

[image]

This reproducibly appears in all those tests, also when running the tests individually.
I pasted the full output of the `Report` button to this [Gist here](https://gist.github.com/HBadertscher/8f636d304ad4fe471341432965fa13fb#file-gistfile1-txt).
It is not the same as you describe, but I assume this is closely related, or what do you think? I must admit that I don't know how I should proceed from here - do you have any ideas what I could check?

This might not be relevant, but after each such crash, the following window is opened:

[image]

The test still runs through and is `PASSED`. I just thought I'd mention it for completeness.

ubruhin, 2018-11-16:
Thanks for your investigation! :smiley: 

It looks like LibrePCB crashes when it is closed right after these tests. But I guess this only happens because LibrePCB is forcefully exited in the test teardown, normally it doesn't exit in such a "hard" way. So I think this is nothing really dangerous, although it would be nice to fix these crashes ;)

On the CI, the tests fail much earlier, not in the teardown. So I'm pretty sure the crashes you encountered are not related to the issue on CI :(

Could you maybe run the test `tests/funq/projecteditor/test_add_component_dialog.py::test_select_device` several times to see if it fails only rarely?

HBadertscher, 2018-11-22:
I can reproduce the fails: they happen in around 2-5 out of 100 runs on my local machine. The following happens: the "Add components" dialog is opened, but somehow Funq can not get the contents of the Symbol Variants dropdown and runs into the timeout in the command:

```
symbvar_items = symbvar_combobox.model_items()
```

(as I'm sure you already know from the CI error messages). From what I've seen, this can happen in all test cases which have that line.

The funny thing is: if I manually click into the Symbol Variants dropdown within the timeout period, then Funq gets the content of the dropdown, the test continues and passes!

Does this help in any way?  Could you give me a hint, how I could  find the reason why this happens? 

Of course we can easily work around this by putting the command into a `try` / `except` and re-trying it if we run into a timeout, but that doesn't really solve the problem.

ubruhin, 2018-11-23:
> I can reproduce the fails: they happen in around 2-5 out of 100 runs on my local machine.

Cool! Unfortunately on Travis-CI it happens much more often :(

> The funny thing is: if I manually click into the Symbol Variants dropdown within the timeout period, then Funq gets the content of the dropdown, the test continues and passes!

Okay, at least this shows us that it must be a bug in Funq, not in LibrePCB :)

> Could you give me a hint, how I could find the reason why this happens?

I took a quick look at the [corresponding code in Funq](https://github.com/parkouss/funq/blob/c3bdb4228b392be587fa7f89776f735fdd81eac4/client/funq/models.py#L751-L763):

```
    def model_items(self):
        """
        Returns the items  (:class:`ModelItems`) associated to this combobox.
        """
        # create and show QComboBoxListView
        self.click()
        # get this QComboBoxListView widget
        internal_qt_name = '::QComboBoxPrivateContainer::QComboBoxListView'
        combo_edit_view = self.client.widget(path=self.path + internal_qt_name)
        model_items = combo_edit_view.model_items()
        # This properly close the QComboBoxListView widget
        combo_edit_view.click()
        return model_items
```

Which looks *very* hacky to me, so I'm not surprised that it doesn't work properly (although I see no reason why it fails only on MacOS) :wink: I think the best option would be to implement that function *properly*, maybe this would solve our issue automatically. If you are interested in this, it would be awesome :) Otherwise probably I could give it a try...

ubruhin, 2019-01-03:
FYI, I started to work on this: https://github.com/parkouss/funq/pull/57