LibrePCB/LibrePCB #369: Copper drawn on top of documentation

dbrgn, 2018-11-06:
##### VERSION / OS / ENVIRONMENT

LibrePCB Version: 0.1.0-rc2
Git Revision:     2800328030285b1c5094975a384a1759c9a4807f
Build Date:       2018-11-04T10:03:33
Qt Version:       5.9.1 (built against 5.9.1)
CPU Architecture: x86_64
Operating System: Arch Linux
Platform Plugin:  xcb

##### SUMMARY

Copper is drawn on top of documentation layer.

[image]

##### STEPS TO REPRODUCE

Package:

```scheme
(librepcb_package e1eb50b1-28ba-4aa8-932d-d8addb5ce9e7
 (name "MP3014")
 (description "The MP3014 package by Luminus.\n\nIt\'s a bit special because the anode pad is smaller than the cathode pad.")
 (keywords "luminus,led,3014,mp3014,1206")
 (author "Danilo B.")
 (version "0.1")
 (created 2018-11-06T20:48:49Z)
 (deprecated false)
 (category da689430-daf5-4a78-8c8d-4231a111eb98)
 (pad ac509123-9f98-4ac0-8b00-16030d5b4e5d (name "1"))
 (pad 86a5c192-c4aa-46d3-ba3d-322855233ae5 (name "2"))
 (footprint 158a3e89-6620-4917-9da7-c8be1bf87b40
  (name "reflow")
  (description "")
  (pad ac509123-9f98-4ac0-8b00-16030d5b4e5d (side top) (shape rect)
   (position -1.45 0.0) (rotation 0.0) (size 1.1 1.0) (drill 0.0)
  )
  (pad 86a5c192-c4aa-46d3-ba3d-322855233ae5 (side top) (shape rect)
   (position 0.475 0.0) (rotation 0.0) (size 1.65 1.0) (drill 0.0)
  )
  (polygon 4f80b7ca-9509-4e8a-9072-273e8525082b (layer top_cu)
   (width 0.0) (fill true) (grab_area true)
   (vertex (position -2.0 1.0) (angle 0.0))
   (vertex (position -1.3 1.0) (angle 0.0))
   (vertex (position -1.3 -1.0) (angle 0.0))
   (vertex (position -2.0 -1.0) (angle 0.0))
   (vertex (position -2.0 1.0) (angle 0.0))
  )
  (polygon d39376a8-7707-4e9b-9c13-077ccb8097e0 (layer top_solder_paste)
   (width 0.0) (fill true) (grab_area true)
   (vertex (position -1.935 0.9) (angle 0.0))
   (vertex (position -1.365 0.9) (angle 0.0))
   (vertex (position -1.365 -0.9) (angle 0.0))
   (vertex (position -1.935 -0.9) (angle 0.0))
   (vertex (position -1.935 0.9) (angle 0.0))
  )
  (polygon 4dbb0034-11f6-4a94-8ef6-57934db740cd (layer top_stop_mask)
   (width 0.0) (fill true) (grab_area true)
   (vertex (position -2.05 1.05) (angle 0.0))
   (vertex (position -1.25 1.05) (angle 0.0))
   (vertex (position -1.25 -1.05) (angle 0.0))
   (vertex (position -2.05 -1.05) (angle 0.0))
   (vertex (position -2.05 1.05) (angle 0.0))
  )
  (polygon b9b2f256-e9d9-4d25-b304-c073dba77d23 (layer top_cu)
   (width 0.0) (fill true) (grab_area true)
   (vertex (position 1.3 0.4) (angle 0.0))
   (vertex (position 2.0 0.4) (angle 0.0))
   (vertex (position 2.0 -0.4) (angle 0.0))
   (vertex (position 1.3 -0.4) (angle 0.0))
   (vertex (position 1.3 0.4) (angle 0.0))
  )
  (polygon 83bd5514-40dd-42b5-8f3c-aa7a1c0b21d4 (layer top_solder_paste)
   (width 0.0) (fill true) (grab_area true)
   (vertex (position 1.3 0.36) (angle 0.0))
   (vertex (position 1.965 0.36) (angle 0.0))
   (vertex (position 1.965 -0.36) (angle 0.0))
   (vertex (position 1.3 -0.36) (angle 0.0))
   (vertex (position 1.3 0.36) (angle 0.0))
  )
  (polygon c8232452-4e08-4233-bdef-90a522b1f1a8 (layer top_stop_mask)
   (width 0.0) (fill true) (grab_area true)
   (vertex (position 1.3 0.45) (angle 0.0))
   (vertex (position 2.05 0.45) (angle 0.0))
   (vertex (position 2.05 -0.45) (angle 0.0))
   (vertex (position 1.3 -0.45) (angle 0.0))
   (vertex (position 1.3 0.45) (angle 0.0))
  )
  (polygon 6dd60106-7400-453f-839a-c0e0cc02fe81 (layer brd_documentation)
   (width 0.127) (fill false) (grab_area true)
   (vertex (position -1.5 0.7) (angle 0.0))
   (vertex (position 1.5 0.7) (angle 0.0))
   (vertex (position 1.5 -0.7) (angle 0.0))
   (vertex (position -1.5 -0.7) (angle 0.0))
   (vertex (position -1.5 0.7) (angle 0.0))
  )
 )
)
```

ubruhin, 2019-07-08:
This is actually caused because you have drawn the polygon on the `brd_documentation` layer instead of `top_documentation` :grin:

So let's close this issue. Could you fix this in the Luminus library? :smiley:

dbrgn, 2019-07-09:
Ah :smile:

dbrgn, 2019-07-09:
Doesn't really make a difference though, even if I choose "Top Documentation" it seems to be shown below the copper:

[image]

[image]

ubruhin, 2019-07-09:
Oops, you're right, sorry :smiley: