LibrePCB/LibrePCB #371: SI_Netline Error

ouabache, 2018-11-07:
##### ACTUAL RESULTS

##### ADDITIONAL INFORMATION

ouabache, 2018-11-07:
Github seems to have eaten my bug report.

What I found was that I was seeing a SI_Netline: Both endpoints are the same frequently when I started a wire on a pin and tried to drop it down on the sheet. It wasn't from all pins and would fail either on vertical or horizontal drops but never both.

I only saw this on parts created by my kicad conversion script. I never saw it from the repository.

I tracked it down to the values used for pin properties. You have a 6 digit mantissa but your library parts all only use 2. My script converts to metric and uses all 6.  That was the trigger. When I limit my script to 2 digits it works.

Here's a test library with my output if you want to try it out.

https://github.com/ouabache/LibrePCB.lplib/archive/master.zip

John Eaton

ubruhin, 2018-11-07:
That's a simple snap grid issue (see https://github.com/LibrePCB/librepcb-rfcs/issues/6). The schematic editor currently only works properly if pins are located *exactly* on the grid. Even if a pin is only 1 nanometer off the grid, snapping doesn't work, that's the issue you have with your converted library.

Basically the schematic editor logic should be adjusted to snap on pins even if they are off the grid.

doom-fr, 2020-05-30:
We agree that this issue disallow to center symbols with even width or height ?
So for now we need to maintained them disaligned. Right ?

And when this issue will be fixed, should we center all already created symbols ?
Will this change be correctly handled by the schematic editor ?

ubruhin, 2020-06-01:
> And when this issue will be fixed, should we center all already created symbols ?

No, IMHO we shouldn't center symbols. I think it is good practice to draw symbols with all pins located on the 2.54mm grid. But nevertheless, the schematic editor should be able to also handle pins off that grid.

Lantcoder, 2021-01-16:
This bug intersects with unavailability to use metric units in components.
[https://librepcb.discourse.group/t/metric-units-in-schematic-components/138/4](url)
What can do?

PR #877 (2021-03-07): Schematic editor: Snap wires to pins/junctions off the grid
When drawing wires in the schematic editor, snap the wire even to symbol pins or junctions not located on the grid. When pressing the SHIFT key while drawing, snapping is temporarily disabled.

Fixes #371.
closed 2021-03-07 by ubruhin