LibrePCB/LibrePCB #372: Cannot find file: ....librepcb-0.1.0_rc2/work/LibrePCB-0.1.0-rc2/libs/googletest/googletest.pro.

Petross404, 2018-11-12:
##### VERSION / OS / ENVIRONMENT
```
Portage 2.3.51 (python 3.7.0-final-0, default/linux/amd64/17.1/hardened, gcc-7.3.0, glibc-2.27-r6, 4.17.4-aufs x86_64)
```
```
LibrePCB Version: sci-electronics/librepcb-0.1.0_rc2
Git Revision:
Qt Version: 5.11.2-r1
Operating System: Linux gentoolinuxserver.com 4.17.4-aufs
```

##### SUMMARY
Make can't file a `.pro` file. I suspect that submodules aren't included in the archives.

##### STEPS TO REPRODUCE
Download a release tar either manually or with portage, unpack it and try to build Librepcb : 
```
qmake && make
```

##### ADDITIONAL INFORMATION
[sci-electronics:librepcb-0.1.0_rc2:20181112-173429.log.txt](https://github.com/LibrePCB/LibrePCB/files/2573017/sci-electronics.librepcb-0.1.0_rc2.20181112-173429.log.txt)

ubruhin, 2018-11-12:
> Download a release tar

What release tar do you mean? The one hosted at GitHub? Unfortunately this one is automatically generated by GitHub and doesn't contain submodules. It's completely useless and I guess I can't delete it :sob: 

You should use our official release ZIP from here, as it contains all needed source files: https://download.librepcb.org/releases/0.1.0-rc2/librepcb-0.1.0-rc2-source.zip

Petross404, 2018-11-13:
> You should use our official release ZIP from here, as it contains all needed source files: https://download.librepcb.org/releases/0.1.0-rc2/librepcb-0.1.0-rc2-source.zip

That worked, thanks.

closed 2018-11-13 by Petross404