LibrePCB/LibrePCB #373: Workspace Settings: Custom PDF viewer

Symbian9, 2018-11-12:
##### SUMMARY

After "PDF Export" LibrePCB try immediately open PDF viewer (default).

##### SOLUTION

Add tab "External applications" to "Workspace Settings" where user could set which PDF viewer/app should be used for open exported PDF.

Here is how such functions shown in *KiCad* app:

[image]

Also add option "Do not open exported PDF".

P.S.: Tab "External applications" in "Workspace Settings" could be later extended for choosing other apps such as *image editor* for opening exported images, *PDF editor*, etc.

ubruhin, 2018-11-12:
Good idea, makes sense :)

I just think that it would be nice to allow choosing "open exported PDF" every time when exporting (maybe sometimes you want to open it, and sometimes not). But I'm not sure how this could be implemented in the user interface... Any ideas?

Beside that user interface thing, I think it shouldn't be too hard to implement this feature, so I'll add the "easy" tag here ;)

ubruhin, 2018-11-12:
Sounds like a good idea! :+1: :smiley: 

For the workspace settings I would suggest following:
- Default value for PDF viewer is an empty string, which means to use the system's default PDF viewer.
- The input field accepts the placeholder "{{PDF}}" which LibrePCB then replaces with the path to the PDF (we could also use a shorter placeholder like "%p" like other applications, but I think for consistency it makes sense to use the same placeholder syntax like everywhere else in LibrePCB).

ubruhin, 2018-11-13:
Looks good! :smiley:

dbrgn, 2018-11-13:
As an improvement idea, we could even detect common PDF viewers on the system and list those as options (with "custom" as a fallback of course).

Some PDF viewers I know and use:

- evince
- mupdf
- zathura

ubruhin, 2018-11-23:
> As an improvement idea, we could even detect common PDF viewers on the system and list those as options (with "custom" as a fallback of course).

I wouldn't do that. It's just more complexity, and as @Symbian9 already wrote, a custom viewer is especially useful for portable viewers which can't be detected anyway. And users which want to use an installed viewer are probably fine with the system's default viewer (otherwise one should consider changing the default viewer system-wide :wink:).

@Symbian9 I would very welcome a pull request to add the workspace settings tab you posted above :)

Symbian9, 2018-11-23:
> I would very welcome a pull request

Well, ... I just *paint* draft above using image editor.

So, there is no code snippet for that :-(

Can you write code for implement this idea?

ubruhin, 2018-11-26:
> Well, ... draft above painted using image editor... 

Ah I see :grin:

> Can you write code for implement this idea?

As this is labeled as "easy", I hope someone else could pick this up :)

Tubbles, 2018-11-26:
> As this is labeled as "easy", I hope someone else could pick this up :)
Well I might be able to pick it up, but I've never done any qt stuff. Is there per chance any learning resources you might recommend @ubruhin? :)

ubruhin, 2018-11-26:
> Well I might be able to pick it up,

Cool! :smile:

> but I've never done any qt stuff. Is there per chance any learning resources you might recommend @ubruhin? :)

The [Qt Core documentation](http://doc.qt.io/qt-5/qtcore-index.html) gives a quick overview about some Qt basics. But generally I would suggest to just start coding and take a look at [Qt's documentation](http://doc.qt.io/qt-5/reference-overview.html) every time you need to know something :wink: In addition, you can just ask me here or in our [IRC](https://webchat.freenode.net/?channels=#librepcb) or [Telegram](https://telegram.me/LibrePCB_dev) chat :)

dbrgn, 2018-11-26:
Oh, and use Qt Creator, UI stuff would be very painful otherwise. With Qt Creator, it should be much easier since you can use the designer :slightly_smiling_face:

NaoPross, 2020-05-01:
Hey, today I felt a bit bored and couldn't concentrate on my normal work so, since I haven't been writing C++ in a while, I gave this one a shot. I've read only the minimal amount of code needed to add this feature so my code may not be the best but I hope it's good enough.

>The input field accepts the placeholder "{{PDF}}" which LibrePCB then replaces with the path to the PDF (we could also use a shorter placeholder like "%p" like other applications, but I think for consistency it makes sense to use the same placeholder syntax like everywhere else in LibrePCB).

Done. I've tested it on Linux (Ubuntu derivate distro) and so far it works, I haven't tested it on other systems though.

>I just think that it would be nice to allow choosing "open exported PDF" every time when exporting (maybe sometimes you want to open it, and sometimes not). But I'm not sure how this could be implemented in the user interface... Any ideas?

I'd suggest a checkbox in the WorkspaceSettingsDialog, that is along the lines of "Ask before opening exported PDFs"

> As an improvement idea, we could even detect common PDF viewers on the system and list those as options (with "custom" as a fallback of course).

I've placed a ComboBox but it only contains "Default" at the moment (and does not affect anything). But thinking about it, trying to detect which PDF viewers are installed could get messy real quick, especially with cross-platform support.

In the UI:
[image]

And in the settings file there are 2 new entries
```scheme
naopross@saturn: v0.2 $ cat settings.lp
(librepcb_workspace_settings
 (user "Nao Pross")
 (application_locale "")
 (default_length_unit millimeters)
 (project_autosave_interval 600)
 (use_opengl false)
 (library_locale_order)
 (library_norm_order)
 (repositories
  (repository "https://api.librepcb.org")
 )
 (use_custom_pdf_reader true)
 (pdf_reader_cmd "evince {{PDF}}")
)
```

Asking for a little feedback before creating a PR.
https://github.com/NaoPross/LibrePCB/tree/add-extapps-settings

ubruhin, 2020-05-02:
Nice! :smiley: 

The new keys in the workspace settings look good so far :+1: But I just realized that the placeholder `{{PDF}}` might not be optimal. Once we add more external applications later, the placeholder would be different for each application. So I wonder if `{{FILEPATH}}` would be better since it is generic enough to use them for *all* external applications?

> I'd suggest a checkbox in the WorkspaceSettingsDialog, that is along the lines of "Ask before opening exported PDFs"

Maybe even provide a choice like "Open exported PDFs: Ask|Always|Never"?

> I've placed a ComboBox but it only contains "Default" at the moment (and does not affect anything). But thinking about it, trying to detect which PDF viewers are installed could get messy real quick, especially with cross-platform support.

Yeah, as I already wrote in a comment above, I'd not implement any auto-detection of PDF readers. I think it's not worth to implement such "black magic" for such a detection which will not be able to detect all readers anyway...

Symbian9, 2020-05-02:
> Maybe even provide a choice like "Open exported PDFs: `Ask` | `Always` | `Never`"?

Yeah!

NaoPross, 2020-05-02:
> The new keys in the workspace settings look good so far +1 But I just realized that the placeholder `{{PDF}}` might not be optimal. Once we add more external applications later, the placeholder [...]

Yeah makes sense, and It's really easy to change so I'll do it.

> Maybe even provide a choice like "Open exported PDFs: Ask|Always|Never"?

That makes most sense to me. Though how do you store enum / multi state settings in your configs? I haven't looked too deep in your code. Should I make an enum or just put a string there? (I.e do you deal with the notorious enum <=> string mapping problem?)
```C++
enum class OpenPDFBehavour : char {
  ASK = 'q', ALWAYS = 'a', NEVER = 'n',
};
...
WorkspaceSettingsItem_GenericValue<OpenPDFBehaviour> openPdfBehaviour;
```
or simply
```C++
WorkspaceSettingsItem_GenericValue<QString> openPdfBehaviour;
```

> Yeah, as I already wrote in a comment above, I'd not implement any auto-detection of PDF readers. I think it's not worth to implement such "black magic" for a detection which will not be able to detect all readers anyway...

Good, I'll get rid of that.

ubruhin, 2020-05-07:
> That makes most sense to me. Though how do you store enum / multi state settings in your configs? I haven't looked too deep in your code. Should I make an enum or just put a string there? (I.e do you deal with the notorious enum <=> string mapping problem?)

You should use an enum and implement the methods `serializeToSExpression()` and `deserializeFromSExpression` for it. See for example the enum `FootprintPad::Shape` [here](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/library/pkg/footprintpad.h). I guess this should also work for workspace settings enums, but if you have any troubles with it, just notify me :)

ubruhin, 2020-05-20:
Implemented in #694 and will be contained in the 0.2.0 release, thanks a lot to @NaoPross! :smiley:

closed 2020-05-20 by ubruhin