LibrePCB/LibrePCB #374: Library Manager: downloaded 3rd-party library from GitHub lost URL info

Symbian9, 2018-11-12:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.0-rc2
Git Revision:     2800328030285b1c5094975a384a1759c9a4807f
Build Date:       2018-11-04T10:03:33
Qt Version:       5.9.1 (built against 5.9.1)
CPU Architecture: x86_64
Operating System: Debian GNU/Linux 9 (stretch)
Platform Plugin:  xcb
```

##### SUMMARY

After import 3rd-party library from GitHub URL info lost (not saved).

##### STEPS TO REPRODUCE

1. Launch LibrePCB;
2. Go to "Library Manager";
3. Switch to "Download manually";
4. Input URL to `master.zip` of 3rd-party library hosted on GitHub;
5. Download this library and switch to installed library details.

##### EXPECTED RESULTS

After import any library by URL, this URL should be shown in "Library Manager".

Also add button "Re-download this library", that should be used for re-install this library from it's URL.

##### ACTUAL RESULTS

URL info lost after 3rd-party library installed.

[image]

Also, impossible update 3rd-party from regular URL, such as `https://github.com/<user>/<3rd-party_library_name>.lplib/archive/master.zip`.

##### ADDITIONAL INFORMATION

- https://github.com/ouabache/LibrePCB_Logic.lplib
  - ZIP: https://github.com/ouabache/LibrePCB_Logic.lplib/archive/master.zip

ubruhin, 2018-11-12:
The URL displayed there is not the URL where the library was downloaded from, it's just some metadata stored directly in the library file: https://github.com/ouabache/LibrePCB_Logic.lplib/blob/master/library.lp#L9

The purpose of that is to see the library URL even if it was downloaded via the API.

> Also add button "Re-download this library", that should be used for re-install this library from it's URL.

This is intended ;) LibrePCB distinguishes between "local" and "remote" libraries. Local libraries can be edited, but remote libraries are read-only. A library downloaded by ZIP is saved as a local library so you can edit it. Then it would be dangerous to re-download it, as local modifications would be lost.

You have following options:
- Clone and update your library manually with Git (as a "local" library).
- Host a (static) json similar to [this one](https://api.librepcb.org/api/v1/libraries/v0.1) somewhere in the internet (or even locally, should also work!) and add the path to it as a new "Repository" in the workspace settings. Then your library can be installed and updated directly with the library manager.
- Contribute your library elements to [the official libraries](https://github.com/LibrePCB-Libraries) :smiley:

Symbian9, 2018-11-12:
> - Contribute your library elements to the official libraries

How send pull request to organization on GitHub? I know that pull request only available for existing repo.

How I could send my repo to https://github.com/LibrePCB-Libraries ?

So, maybe, you will create `Contribute` repo as issues tracker for contributing library discussion?

- https://github.com/LibrePCB-Libraries/Contirbute/issues

Symbian9, 2018-11-12:
> You can clone one of the existing libraries to ~/LibrePCB/v0.1/libraries/local/, do modifications and then send a pull request with your changes to the upstream repository

Okay, you tell about 'modifications' (forking) existed libraries on https://github.com/LibrePCB-Libraries

But if I will create new library from scratch, how pull it to https://github.com/LibrePCB-Libraries ?

dbrgn, 2018-11-12:
If you need a new library, let us know and we can create an empty library. You can then create pull requests adding components!

Symbian9, 2018-11-13:
> If you need a new library, let us know

Above I already propose:

> So, maybe, you will create `Contribute` repo as issues tracker for contributing library discussion?
>
> - https://github.com/LibrePCB-Libraries/Contirbute/issues

Symbian9, 2018-11-13:
> I will take a look at your library soon, to decide then how to proceed.

FTR, https://github.com/ouabache/LibrePCB_Logic.lplib is NOT my library.

closed 2018-11-13 by Symbian9

rnestler, 2018-11-13:
> Contribute your library elements to the official libraries smiley

Do we document somewhere, how to contribute to the official libraries? Especially when somebody want's to create a new library?