LibrePCB/LibrePCB #385: Unbundle statically linked libraries

arunisaac, 2018-12-01:
##### SUMMARY
I'm packaging LibrePCB for Guix, a GNU/Linux distribution. LibrePCB has several libraries bundled into its source under `libs/`. This makes packaging less efficient with dependencies duplicated across several projects. It also causes dependencies used in LibrePCB to lag behind their latest released version.

##### SOLUTION
Unbundle all dependencies, and link to them as dynamic libraries. Guix and most other GNU/Linux distributions prefer to deal with dependencies as separate packages. This lets the distro's package manager handle dependency management and avoids the problems I mentioned earlier.

Thank you for your consideration!

ubruhin, 2018-12-01:
Hi @arunisaac,

I agree that for many Linux distributions it would make sense to just link to dynamic libraries instead of using our bundled, statically linked libraries. The main reason why we link statically is that it heavily depends on the distribution which dynamic libraries are available or not. For example on Windows we have to provide all dependencies by ourselves. So we would need to allow selecting which libraries should be statically linked, and which dynamically.

At least there are very few and small libraries which could be dynamically linked. Probably only hoedown and quazip on most distributions(?) googletest also, but it is not needed for the application, only for tests.

Generally I'm open for pull requests adding some kind of switches to select which libraries should be dynamically linked. But linking statically should stay the default, otherwise we will get issues on systems which do not provide dynamic libraries ;)

arunisaac, 2018-12-01:
I am fine with adding switches to select which libraries are to be
dynamically linked. That would be sufficient for Guix's purposes.

I noticed that many of these dependencies (googletest, hoedown, quazip,
etc.) have been forked and patched for use with LibrePCB. Will LibrePCB
build successfully against the unmodified dependencies?

And, some dependencies (delaunay-triangulation, parseagle, sexpresso,
etc.) don't yet have release tarballs. Dynamic linking and packaging
would be easier if we had specific release tarballs to work with.

ubruhin, 2018-12-01:
> I noticed that many of these dependencies (googletest, hoedown, quazip, etc.) have been forked and patched for use with LibrePCB. Will LibrePCB build successfully against the unmodified dependencies?

As far as I remember, they are only patched to add a qmake project file. So these patches aren't needed if you dynamically link to that libraries.

> And, some dependencies (delaunay-triangulation, parseagle, sexpresso, etc.) don't yet have release tarballs. Dynamic linking and packaging would be easier if we had specific release tarballs to work with.

I don't know about packaging of delaunay-triangulation and sexpresso, but parseagle and fontobene are made by ourselves especially for LibrePCB. They aren't packaged and distributed separately, so generally I would suggest to consider them just as part of LibrePCB and thus keep statically linking to them. IMHO packaging them separately leads to additional effort and risk, without any advantages.

rnestler, 2019-01-09:
> fontobene are made by ourselves especially for LibrePCB. They aren't packaged and distributed separately, so generally I would suggest to consider them just as part of LibrePCB and thus keep [...]

Actually if we want to encourage other people to use fontobene as a stroke font library, it would be an advantage to support packaging them as distro packages and also to provide releases.

ktrace, 2019-01-23:
I just want to make same ticket. Yes, I agree, at least googletest and quazip must be linked dynamically.

dbrgn, 2020-05-12:
I tried to find an Arch TU (Trusted User) to maybe create an official ArchLinux package. Someone (dvzrv) signaled interest in maintaining an Arch community package. However, people in IRC also objected the vendoring of dependencies.

I see that vendoring makes development easier but it also hinders packaging in distros. We should at least make it possible to patch out the static linking / vendoring, even if it's not the standard way of building.

Right now we use the following libs:

- LibrePCB Custom:
  - librepcb
- Standalone but made by LibrePCB as well:
  - [fontobene-qt5](https://github.com/fontobene/fontobene-qt5/tree/f167099fa14659f616fe543d6f8f6baf45f286cc)
  - parseagle
- With custom patches:
  - delaunay-triangulation
  - muparser
  - sexpresso (compilation fix that could be upstramed)
  - type_safe (commented out two operators that might be dangerous)
- Without modifications (except for qmake project files):
  - [clipper](http://www.angusj.com/delphi/clipper.php)
  - googletest (for unit tests only)
  - [hoedown](https://github.com/hoedown/hoedown)
  - [optional](https://github.com/TartanLlama/optional)
  - [quazip](https://github.com/stachenov/quazip/)

I would suggest the following first steps:

- Create a custom AUR package for fontobene-qt5 (simple since we control it ourselves)
- Allow optional dynamic linking to quazip (since that library is already packaged for Arch in the "extra" repo)

@ubruhin what steps would I need to take to do optional dynamic linking of a library in qmake?

FFY00, 2020-05-12:
Yes, vendored depdendencies are blocking us from pushing a package. Once this gets resolved I would gladly work on it.

ubruhin, 2020-05-14:
> what steps would I need to take to do optional dynamic linking of a library in qmake?

Hmm I don't know exactly, but I assume it's quite a pain due to the many different qmake files with a lot of references to the libraries :wink: Actually I was never happy with our qmake project file structure (IMHO it's too complicated), but I don't have the qmake knowledge to do a major refactoring...

dbrgn, 2020-05-14:
Yeah, first successful build of LibrePCB as a dynamically linked binary that links to liblibrepcbcommon.so, which in turn links to system quazip 🎉

[image]

The general approach is as follows:

- By default, everything is built as a static binary using vendored libs
- You can choose to unbundle certain libraries by specifying `UNBUNDLE+=quazip` as a qmake parameter
- If any library is unbundled, this will lead to dynamic builds instead of static builds
- Some libs like quazip require different imports depending on the header source. In those cases, I conditionally define a `SYSTEM_QUAZIP` preprocessor variable which can then be used to import the correct headers

FedericoCeratto, 2020-05-15:
A clarification: in most distributions, unbundling libraries does not requires dynamic linking. Library sources packages can be pulled in as bulild dependencies if needed.

dbrgn, 2020-05-15:
> A clarification: in most distributions, unbundling libraries does not requires dynamic linking.

Yes, but one of the reasons distributions ship system libraries instead of providing static binaries for everything is that total resource requirements can be reduced by sharing those libraries between different programs. So when unbundling (especially in the context of distro packaging), it probably makes sense to link dynamically as well, right?

FFY00, 2020-05-15:
> A clarification: in most distributions, unbundling libraries does not requires dynamic linking. Library sources packages can be pulled in as bulild dependencies if needed.

That is totally wrong, most distributions require dynamic linking. There might be distributions that don't require this but it is definitely not the norm.

FedericoCeratto, 2020-05-15:
@dbrgn generally speaking, yes. The best thing about dynamic linking is that allows fast security updates on vulnerable libraries without having to rebuild, test and distribute their consumers.

dvzrv, 2020-05-28:
@dbrgn Thanks for the work on this! It is much appreciated.

I would like to package librepcb for Arch Linux, but the amount of vendored libraries is a bit scary. After your pull request this will be much more straight forward for downstreams to use! :)

dbrgn, 2020-05-28:
Thanks for the feedback @dvzrv. Note that @FFY00 also seems to be working on an Arch package. Maybe you two can coordinate :slightly_smiling_face: 

I hope to finish the unbundling work tomorrow (at least for quazip and fontobene-qt5, for now).

closed 2020-06-09 by ubruhin

itviewer, 2020-08-13:
It is amazing to statically link all dependencies. When you develop, you only modify one line of code and it takes a long time to compile to see the result.
Why do we have to comply with the Linux installation file location standard?
Why can't we create a `.run ` file like the windows installer?

dvzrv, 2020-08-13:
@itviewer I'm not entirely sure if you are joking or whether this is a serious question.

In case it is a serious question, you are answering it yourself: It ([FHS](https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard)) is a (relatively) well-defined **standard**.
If you would like to install things using a script that "probably" works somewhere, that is up to you (after all you can do whatever you want to your own machine). It is just not anything that will happen in a Linux distribution.

Rather on-topic: In case you are interested in reading up on why bundled libraries are usually not a good idea, please have a look at the excellent [writeup by Fedora](https://fedoraproject.org/wiki/Bundled_Libraries?rd=Packaging:Bundled_Libraries) on that topic.

dbrgn, 2020-10-05:
With the [0.1.5 release](https://librepcb.org/blog/2020-10-04_release_0.1.5/), the following three libraries can now be unbundled:

- quazip
- polyclipping
- fontobene-qt5

With the PR at #779, hoedown will be gone as well in the next version, as long as the project is being built on Qt >=5.14.

More information about dynamic linking / unbundling can be found at https://developers.librepcb.org/d5/d96/doc_building.html#doc_building_unbundling.

dbrgn, 2020-10-16:
Here's a document with the current unbundling status (will be updated from time to time): https://md.coredump.ch/TDSZgzLBSTG6a6l-KbowsA