LibrePCB/LibrePCB #387: Crash when checking "Required" checkbox in component signal

eduardosm, 2018-12-04:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.0-unstable
Git Revision:     f08fe1f6a4dd07355b0dedcbabb20abcd0aaf8b1
Build Date:       2018-12-04T17:54:25
Qt Version:       5.11.2 (built against 5.11.2)
CPU Architecture: x86_64
Operating System: Manjaro Linux
Platform Plugin:  xcb
```
Tested too in 0.1.0-rc2

##### SUMMARY
Crash when checking "Required" checkbox in component signal

##### STEPS TO REPRODUCE
1. Open any library with components
2. Open a component
3. Check the "Required" checkbox in two signals (uncheck them first if they are all checked)

##### EXPECTED RESULTS
The checkbox is checked.

##### ACTUAL RESULTS
LibrePCB crashes

##### ADDITIONAL INFORMATION
[Backtrace](https://gist.github.com/eduardosm/ba2d24cd3d915f418a8d01eb3c21a70c)

ubruhin, 2018-12-04:
Hmm I can't see something strange in the stacktrace...

Could you try to reproduce it with the official binaries (e.g. the AppImage)?

eduardosm, 2018-12-05:
The appimage seems to work well. I tried to look into it, but I couldn't find any clue either. I suspect it has something to do with my qt version

ubruhin, 2018-12-05:
Ok, good to know that the stable release is stable :grin:

Did you already try a completely clean rebuild? I also have very strange crashes from time to time, I suspect these are caused by an issue of the build system (an object file is not rebuilt when it should). A rebuild then fixes these crashes...

eduardosm, 2018-12-05:
Yes, in fact, I had to rebuild in order to have debug symbols (erase the `build` directory and run `qmake` again)

ubruhin, 2018-12-05:
Perfect :+1:

closed 2018-12-05 by ubruhin